OpenEtherCATsociety/SOEM #695: Understanding the timing of the Linux sync point

brentonchris, 2023-04-12:
in this issue “some problems with "the linux sync point 50us later than DC sync" #519”

My understanding is that Linux sync point is a relative time, which is relative to the reference clock of the first slave station. This time should be later than sync0. If the DC time is 1ms and I set the Linux sync point to 300us, the master station will organize data frames and send them out at the DC cycle of 300us, and the DC shift time is 700us. If I set the Linux sync point to -300us, then the master station will organize data frames and send them out at the DC cycle of 700us, and the DC shift time at this time is 300us?

I don't know if my understanding is correct?

ArthurKetels, 2023-04-12:
Not quite. The linux sync point is the time the master starts transmitting the PDO packet(s) relative to the DC sync0/1 event (- is before, + is after). DC shift time is slave internal and determines when (1) the outputs are valid after sync0/1 and (2) the inputs are available after sync0/1. Due to packet transmit times and linux network stack latency the linux sync point has no absolute relation to the DC shift time.

ArthurKetels, 2023-04-12:
1, The following link explains it quite nicely : https://infosys.beckhoff.com/english.php?content=../content/1033/ethercatsystem/2469120395.html&id=

2 and 3. It is not as simple as you might think. To establish the correct trigger time of master and each slave you need to lay out your timing requirements next to the operational limits of your combined system. You have to take into account your master HW and OS performance with respect to jitter and latency. Not only as a single number but also the probability distribution. Then your slave topology. A large number of slaves will cause non trivial propagation delays. It is best to lays this all out in a spreadsheet so you can see the effects of timing variations, propagation delays and trigger shift times.

Then there is the possibility you do not know all these effects and don't care because the your application is not time critical and any setting will work satisfactory.

SOEM is just a tool, you are the master carpenter and responsible for the end result. And yeah, a master carpenter needs skills.

brentonchris, 2023-04-13:
Thanks for reply.

We are a company that specializes in motion control. Currently, we can connect up to 32 servo drives with a DC time of 1ms. We would like to find a way to determine the shift time to ensure it is appropriate. We are not sure if there are any tools available to help test this time.

Can you leave your email if it's convenient.
My email:brenton@163.com

brentonchris, 2023-08-10:
Thank you for your reply. This issue has been clarified.

closed 2023-08-10 by brentonchris