OpenEtherCATsociety/SOEM #696: Overlapping and legacy process data devices in same network

d-loret, 2023-04-15:
I am having difficulties getting an Elmo Platinum motor controller to communicate in the same EtherCAT network as devices that use legacy mapping (Beckhoff EL3602, EL2124, etc.). When I request Operational state to all slaves, they do not transition to it. They remain in Safe-Operational.

I am able to communicate with the Platinum if there are no other devices with it (by setting `blockLRW` in the Platinum to 1 during initialization AND not calling the overlap version of send_processdata, or viceversa). Likewise, I am able to communicate with the rest of my devices if the Platinum is not present (by using `ecx_send_processdata` and not fiddling with `blockLRW`).

How can I communicate when all the devices are present? I tried using groups and calling `ecx_send_overlap_processdata_group` on the group that has the Platinum, and `ecx_send_processdata_group` on the group that has all other devices (and not fiddling with `blockLRW`). But I have not been successful, so I am missing something.

A high level overview of the functions I call during initialization is the following:

```
#define NONOVERLAP_GROUP = 1
#define OVERLAP_GROUP = 2
...
my_ecx_context.grouplist = (ec_groupt*)calloc(1, sizeof(ec_groupt[3]))
my_ecx_context.maxgroup = 3;
...
ecx_init(&my_ecx_context, my_interface);
ecx_config_init(&my_ecx_context, FALSE);
...
// I assign some properties of the slaves in my_ecx_context.slavelist such as as function pointer PO2SOconfigx.

// For each slave slave_idx that is not a platinum
//     my_ecx_context.slavelist[slave_idx].group = NONOVERLAP_GROUP
// For each slave slave_idx that is a Platinum
//     my_ecx_context.slavelist[slave_idx].group = OVERLAP_GROUP

int iomap_size = ecx_config_map_group(&my_ecx_context, &my_IOmap, NONOVERLAP_GROUP);
int iomap_size_overlap = ecx_config_map_group(&my_ecx_context, &my_overlap_IOmap, OVERLAP_GROUP);

ecx_statecheck(&my_ecx_context, 0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE);
...
ecx_readstate(&my_ecx_context);
...
// Execute the following instructions until state is Operational or a maximum number of attempts have been made.
ecx_send_processdata_group(&my_ecx_context, NONOVERLAP_GROUP);
ecx_receive_processdata_group(&my_ecx_context, NONOVERLAP_GROUP, EC_TIMEOUTRET);

ecx_send_overlap_processdata_group(&my_ecx_context, OVERLAP_GROUP);
ecx_receive_processdata_group(&my_ecx_context, OVERLAP_GROUP, EC_TIMEOUT_RET);

my_ecx_context.slavelist[0].state = EC_STATE_OPERATIONAL;
ecx_writestate(&my_ecx_context, 0);

ec_state actual_state = ecx_statecheck(&my_ecx_context, 0, state, EC_TIMEOUTSTATE);
```

But `actual_state` above never becomes `EC_STATE_OPERATIONAL`. If I call `ecx_readstate` and print the state and `Altstatuscode` of each slave, I get the following:

```
slave[1], Actual state = EC_STATE_SAFE_OP, StatusCode=   0 : No error
slave[2], Actual state = EC_STATE_SAFE_OP, StatusCode=  1b : Sync manager watchdog
slave[3], Actual state = EC_STATE_SAFE_OP, StatusCode=   0 : No error
slave[4], Actual state = EC_STATE_SAFE_OP, StatusCode=   0 : No error
``` 
"slave[2]" above is the Beckhoff EL3602 which only has inputs ("slave[4]" is the Platinum).

I also see that return values (i.e. wkc) of the calls to `ecx_receive_processdata_group` change with each attempt. Wkc from the Platinum group is usually -1 during the first attempts and 3 toward the last attempts. For the other devices, sometimes it is 2, and sometimes 3. Sometimes the wkc of both the Platinum and the other devices is 3. However, the states and `Altstatuscode` are always the same.

What am I doing wrong? This is the first time we are incorporating groups and the send_overlap_processdata function in our code, but I am not even sure if this is the right approach to get all devices communicating properly in the same network.

I would really appreciate some guidance or tips on this issue.

nakarlsson, 2023-04-15:
Use only overlapping. Legacy is the legacy SOEM way of mapping data. 
Elmo use the TI ESC that require overlapping, hence you should use it for the entire network.

d-loret, 2023-04-18:
@nakarlsson, when using overlapping for the entire network I encounter a similar problem when all devices are in the same network. When trying to enter Operational, they stay in Safe-Operational. The slight difference is that for the Beckhoff 3602 (analog input) and the Platinum I get a "Sync manager watchdog" status. For the EK1100 and EL2124 I get "No error."

Likewise, if I disconnect the Platinum from the EK1100, the EL3602 and the EL2124 work fine. If I connect directly to the Platinum it also works fine. But when all are in the same network, I get the aforementioned issues.

nakarlsson, 2023-04-18:
Can you share the "application" again? I suggest you revert it to a basic configuration, no groups, no bock LRW BUT use Overlapping.  Also, post the slaveinfo console output of the setup using overlapping..

Post a wireshark of a startup including the attempts to go to OP (slaveinfo doesn't do that)

ArthurKetels, 2023-04-18:
I know the Elmo drives are relatively slow in state transition. They take longer than the PDO watchdog time out. So it is very important that a constant stream (at the correct time interval) of PDO cycles is provided during the state transition pre-OP -> OP. This should be asynchronous from the state transition check (and therefore require at least two tasks). The code flow of simple_test program is NOT suitable.

d-loret, 2023-04-19:
I changed somewhat the order of the instructions in my snippet above, and all devices are now achieving Operational state. Before we had something along the lines (after replacing for send_overlap_processdata):

```
int attempt = 0;
while (true) {
  ecx_send_overlap_processdata(&my_ecx_context);
  ecx_receive_processdata(&my_ecx_context);

  my_ecx_context.slavelist[0].state = EC_STATE_OPERATIONAL;
  ecx_writestate(&my_ecx_context, 0);

  ec_state actual_state = ecx_statecheck(&my_ecx_context, 0, state, EC_TIMEOUTSTATE);

  ++attempt;

  if (actual_state == EC_STATE_OPERATIONAL) {
    break; // Good to go
  } else if (attempt > max_attempts) {
    return false; // Signal error to the caller of the initialization function
  }
}
```

And I moved the writing of state where `simple_test` has it:

```
my_ecx_context.slavelist[0].state = EC_STATE_OPERATIONAL;

ecx_send_overlap_processdata(&my_ecx_context);
ecx_receive_processdata(&my_ecx_context);

ecx_writestate(&my_ecx_context, 0);

int attempt = 0;
while (true) {
  ecx_send_overlap_processdata(&my_ecx_context);
  ecx_receive_processdata(&my_ecx_context);

  ec_state actual_state = ecx_statecheck(&my_ecx_context, 0, state, EC_TIMEOUTSTATE);

  ++attempt;

  if (actual_state == EC_STATE_OPERATIONAL) {
    break; // Good to go
  } else if (attempt > max_attempts) {
    return false; // Signal error to the caller of the initialization function
  }
}
```
Since the original order has been working for us with all other devices including Elmo Gold drives, I am not sure why it is making a difference now for the Platinum. But with the latter order the Platinum is now able to share the network with the other devices.

ArthurKetels, 2023-04-19:
Two problems in your original startup code:
1 - Each time you execute ecx_writestate() the slave gets triggered that a state change is ordered. Although the state remains the same it could alter the slave behaviour (amazing how lousy some slave firmware is written).
2 - ecx_statecheck has EC_TIMEOUTSTATE as time value (default 2ms). In your code this is the effective cycle rate as long not all slaves have transitioned to OP. It is best to have the same cycle rate as in your OP state. Adjust your timeout value to that.

And more in general it is far better to have separate tasks for PDO cycle and state management.

In pseudo code:
```
TASK PDO
{
   while(runPDO)
   {
      exc_send_overlap_processdata()
      exc_receive_processdata()
      if (WKC == expectedWKC)
      {
         ..do PDO input and output stuff..
      }
      ..calculate next absolute cycle start time (either freerun or synchronous to DC)
      ..wait until next cycle start time..
   }
}

TASK MAIN
{
   runPDO = FALSE
   ..init stuff..
   ecx_init()
    ecx_config_init()
    ...
    ecx_config_map()
    ...
   ecx_config_dc()
   runPDO= TRUE
   ecx_statecheck(EC_STATE_SAFE_OP)
   while(!PDOsynchronized) { ..wait.. }
   ecx_writestate(EC_STATE_OP)
   while(program_runs)
   {
      ecx_statecheck(EC_STATE_OP)
      if (!EC_STATE_OP)
      {
          ..handle error state..
      }
   }
}
```

nakarlsson, 2023-10-25:
question completed

closed 2023-10-25 by nakarlsson