OpenEtherCATsociety/SOEM #697: Dear leader, may I ask if this project will continue to be developed using QT?

mygod520, 2023-04-15:
1.这个项目我看了很久，想用qt来移植它并使用他，弄了2个晚上，可是结果确不尽人意。
2.在win 平台上，只能用VS来写吗，是否可以出一个相关教程

mygod520, 2023-04-15:
PS E:\YANKONG\SOEM\build> mingw32-make.exe
[  4%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.obj
gcc.exe: error: /D: No such file or directory
gcc.exe: error: _CRT_SECURE_NO_WARNINGS: No such file or directory
gcc.exe: error: /WX: No such file or directory
CMakeFiles\soem.dir\build.make:75: recipe for target 'CMakeFiles/soem.dir/soem/ethercatbase.c.obj' failed
mingw32-make[2]: *** [CMakeFiles/soem.dir/soem/ethercatbase.c.obj] Error 1
CMakeFiles\Makefile2:152: recipe for target 'CMakeFiles/soem.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/soem.dir/all] Error 2
Makefile:134: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

**### Compilation error in 'MinGW Makefiles'；**

ArthurKetels, 2023-04-16:
If you want to compile with mingw please see #513. I would recommend to use Linux. On Windows only Visual Studio C is supported. Performance and latency is a magnitude better on Linux though. 

SOEM does not care what you use as GUI. It is up to you to integrate SOEM with your application and GUI. 

P.s. Do not assume all people in the world understand and read Mandarin. This project is English only.