OpenEtherCATsociety/SOEM #698: machine1 can OP run，machine2 cant OP，same code

zouzhe1, 2023-04-18:
# 1 introduce
My program runs in **Raspberry Pi**, and the motor can rotate normally.

But if the same program is placed in **X86** or **Jetson**, the motor will not rotate and an error message will be reported as follows.

I think this issue may be related to time.

My servo is the SV630N of inovance，**only 1 servo**

# 2，The error is as follows:

```bash
SOEM (Simple Open EtherCAT Master)
Redundancy test
# /dev/cpu_dma_latency set to 0us
Starting Redundant test,mode=8
ec_init on eth0 succeeded.
1 slaves found and configured.
BOBO2 Drive setup
Slaves mapped, state to SAFE_OP.
Slave:1 Name:InoSV630N Output size:152bits Input size:200bits State: 4 delay:0.1
Calculated workcounter 3
Request operational state for all slaves
Not all slaves reached operational state.
Slave 1 State=0x14 StatusCode=0x0027 : Freerun not supported

Request safe operational state for all slaves
End driver test, close socket
End program
```

# 3，error wireshark file  is here:
[PCAPfile230418.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11257562/PCAPfile230418.zip)

# 4，slaveinfo
 ```bash
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:InoSV630N
 Output size: 152bits
 Input size: 200bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00100000 ID: 000c0112 Rev: 00010000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  19 F:00010064 Type:3
 SM3 A:1c00 L:  25 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  19 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000013 Ll:  25 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 06
 CoE details: 0d FoE details: 00 EoE details: 02 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

zouzhe1, 2023-04-18:
@ArthurKetels 
You're great. Based on the PCAP file, you can see such a detailed problem. 
I don't even know how to look at Wireshark to find the problem.

This problem stuck with me for a long time, Thank you for taking a look at my program.

Here is my program：
```c
void test_driver(char *ifname, int mode)
{
   int cnt, i, j ;
   BO2_Drive_Inputs *iptr;
   BO2_Drive_Outputs *optr;
   struct sched_param schedp;
   cpu_set_t mask;
   pthread_t thread;
   int ht;
   int chk = 2000;
   int64 cycletime;
   struct timespec ts, tnow;

   CPU_ZERO(&mask);
   CPU_SET(2, &mask);
   thread = pthread_self();
   pthread_setaffinity_np(thread, sizeof(mask), &mask);

   memset(&schedp, 0, sizeof(schedp));
   schedp.sched_priority = 99; /* 设置优先级为99，即RT */
   sched_setscheduler(0, SCHED_FIFO, &schedp);

   printf("Starting Redundant test,mode=%d\n",mode);

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n", ifname);
      /* find and auto-config slaves */
      if (ec_config_init(FALSE) > 0)
      {
         printf("%d slaves found and configured.\n", ec_slavecount);
         /* wait for all slaves to reach SAFE_OP state */

         int slave_ix;
         for (slave_ix = 1; slave_ix <= ec_slavecount; slave_ix++)
         {
            ec_slavet *slave = &ec_slave[slave_ix];
            slave->PO2SOconfig = bobo2_drive_setup;
         }

         /* configure DC options for every DC capable slave found in the list */
         ec_config_map(&IOmap); // 此处调用drive_setup函数，进行PDO映射表设置
         ec_configdc(); // 设置同步时钟，该函数必须在设置pdo映射之后；

         // setup dc for devices
         for (slave_ix = 1; slave_ix <= ec_slavecount; slave_ix++)
         {
            ec_dcsync0(slave_ix, TRUE, 4000000U, 20000U);//循环时间，偏移时间
            //ec_dcsync01(slave_ix, TRUE, 4000000U, 8000000U, 20000U);
         }

         printf("Slaves mapped, state to SAFE_OP.\n");
         ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE);

         /* read indevidual slave state and store in ec_slave[] */
         ec_readstate();
         for (cnt = 1; cnt <= ec_slavecount; cnt++)
         {
            printf("Slave:%d Name:%s Output size:%3dbits Input size:%3dbits State:%2d delay:%d.%d\n",
                   cnt, ec_slave[cnt].name, ec_slave[cnt].Obits, ec_slave[cnt].Ibits,
                   ec_slave[cnt].state, (int)ec_slave[cnt].pdelay, ec_slave[cnt].hasdc);
[... 202 lines cut ...]
   }
}

/* PI calculation to get linux time synced to DC time */
void ec_sync(int64 reftime, int64 cycletime, int64 *offsettime)
{
   static int64 integral = 0;
   int64 delta;
   /* set linux sync point 50us later than DC sync, just as example */
   delta = (reftime - 50000) % cycletime;
   if (delta > (cycletime / 2))
   {
      delta = delta - cycletime;
   }
   if (delta > 0)
   {
      integral++;
   }
   if (delta < 0)
   {
      integral--;
   }
   *offsettime = -(delta / 100) - (integral / 20);
}
```

zouzhe1, 2023-04-18:
@ArthurKetels 
When running, my drive will also report an error, and the query manual says: no synchronization signal, the mechanism of generation: When the servo communication switches to the OP state, the MCU does not receive the synchronization signal.

zouzhe1, 2023-04-20:
@ArthurKetels 
Hello, can you help me take a look?
 This problem has been holding me back for half a year and I haven't figured out how to solve it. 
Is that the sentence 'ec'_ dcsync0(slave_ix, TRUE, 4000000U, 20000U); ？
How much should the cycle time of dcsync0 and the cycle time of my tes function be set separately?
 Are they related? 
Thank you very much! Thank you very much!

Liwhitehead666, 2023-11-24:
bro，I have same problem insv660n. Are you solved this now. IfThank you very much!

Liwhitehead666, 2023-11-24:
bro，I have same problem insv660n. Are you solved this now. Thank you very much!

Liwhitehead666, 2023-11-24:
hello，are you solved the“freerun” problem，can we talk about this. I just stucked here in the sv660n.