OpenEtherCATsociety/SOEM #703: Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog

RyuAmakaze, 2023-05-02:
Although it temporarily transitions to OP, it becomes SAFE OP+ERROR, 
and the ecatcheck function in a separate thread repeats the synchronization between OP and SAFE OP+ERROR.
After checking issues such as [issue456](https://github.com/OpenEtherCATsociety/SOEM/issues/456), I suspected a synchronization problem and tried running void *ecatthread( void *ptr ) in a separate thread with reference to [red_test.c](https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/linux/red_test/red_test.c).
It is unclear whether this is the cause, but ec_DCtime remains at 0 and the error message remains the same. 

Additionally, I tried the following code, but I am unsure if the values within the ec_dcsync function are correct.
```
ec_dcsync0(i, TRUE, 125 * 1000, 125 * 1000);
```

We have already completed the PDO mapping and have received a response from the manufacturer that it is correct.

### Question
・ec_DCtime = 0 is problem ?
・Is it correct behavior that the working counter(wkc) goes to 3 or 1?

### Enviroment
Jetson ORIN NX
NX215 (carrier board)
ROS1 C++

### ESI File
[ORIENTALMOTOR_AZDxA-KED_rev0301](https://github.com/OpenEtherCATsociety/SOEM/files/11371108/ORIENTALMOTOR_AZDxA-KED_rev0301.zip)

### wireshark log
[wireshark_log.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11371234/wireshark_log2.zip)

### slaveinfo.c output 
[slaveinfo.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11370533/slaveinfo.zip)

ArthurKetels, 2023-05-03:
Your application program does not follow the normal flow that is accepted to get stable communication. First of all you do not invoke ec_config_dc(). So there will be no clock distribution. Then the PDO exchange is once every few seconds and very irregular. This needs to be constant at a fixed interval (for example once every millisecond).

RyuAmakaze, 2023-05-08:
Thanks [ArthurKetels](https://github.com/ArthurKetels).
A new simple test was created to confirm stable operation, 
and the ec_sync function taken from redtest played an important role.

[simpletest.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11417465/simpletest.zip)

closed 2023-05-08 by RyuAmakaze