OpenEtherCATsociety/SOEM #705: How to deal with Potential Supply terminals and addressing

MarcoMuellner, 2023-05-04:
Currently, i have the following setup:

1) EK1100
2) EL2004
3) EL2004
4) EL2612
5) **EL9100**
6) EL2004
7) ...
Everything up to the EL9100 is addressed correctly, and SOEM ignores the EL9100 in the addressing. Here is the output of slaveinfo:

```
Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.1.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0

Slave:2
 Name:EL2004
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 145[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 07d43052 Rev: 00120000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:3 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0

Slave:3
 Name:EL2004
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 285[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1003
 Man: 00000002 ID: 07d43052 Rev: 00120000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   1 Lsb:4 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0

Slave:4
 Name:EL2612
 Output size: 2bits
 Input size: 0bits
 State: 4
 Delay: 425[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1004
 Man: 00000002 ID: 0a343052 Rev: 00100000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000001 Ll:   1 Lsb:0 Leb:1 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 150[mA]
 only LRD/LWR:0

Slave:5
 Name:EL2004
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 570[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1005
 Man: 00000002 ID: 07d43052 Rev: 00120000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000001 Ll:   1 Lsb:2 Leb:5 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0
```

However, when i try to address slave 5 (in this case the EL2004, physically it is after the EL9100, so in this case it would be slave 6) it addresses the Outputs of slave 4. Is there a smart way to deal with these types of slaves?

Thanks for any help :)

nakarlsson, 2023-05-04:
Don't treat the EL9100 as a slave?

From manual,

6.1 Configuration with TwinCAT System Manager -
passive terminals
EL9011, EL9070, EL9080;
EL9100, EL9150, EL9181, EL9182, EL9183, EL9184, EL9186, EL9187, EL9188, EL9189, EL9190;
EL9200, EL9250, EL9290;
EL9400;
EL9540, EL9550;
EL9570

There are no address or configuration settings necessary for the mentioned passive terminals

ArthurKetels, 2023-05-04:
And look at the logical memory mapping.
Slave 4 : logical address 0x0001  bits 0 to 1
Slave 5 : logical address 0x0001 bits 2 to 5
So yes, if you manipulate byte 0x0001 in the IO mapping you will touch both slave 4 and 5. Use proper bit manipulation to set and clear each output. Slaves with output sizes below 8 bits are automatically concatenated as long they do not pass byte boundaries. This increases packet and storage efficiency.

MarcoMuellner, 2023-05-04:
I don't think i understand this at all.

> There are no address or configuration settings necessary for the mentioned passive terminals

Yes thats what i assumed and what i read in the documentation as well. But then the adressing doesn't seem to work. Let me share some code, to illustrate what i mean. Say i have some code like this:

```c
void SetOutputs(int slave, int index, int value)
{
        uint8_t val = (1 << index);
        uint8_t *data_ptr = ec_slave[slave].outputs;

        if (value > 0) {
            val = *(data_ptr) | val;
        } else {
            val = ~val;
            val = *(data_ptr) & val;
        }
        *(data_ptr) = val;
}
```

this works perfectly fine with all slaves until slave number 4, but just fails to address the correct outputs after the EL9100, and that just seems weird to me. What am i missing here? Any help is appreciated :)

ArthurKetels, 2023-05-04:
It is trivial to check the content of your IOmap by printing it out on screen. It is only 2 bytes. Then you see if the correct bits are set or cleared. If that is OK you can check again the content of the LRW packet on the wire with Wireshark. You should see the same bit set in the packet at the second byte.

MarcoMuellner, 2023-05-05:
Added some logging, i think i understand the issue now:

```
Slave: 0 has no outputs
Slave: 1 has no outputs
Slave: 2, Output data address: 5674B460
Slave: 2, Output states:
Byte 0: 00000000
Slave: 3, Output data address: 5674B460
Slave: 3, Output states:
Byte 0: 00000000
Slave: 4, Output data address: 5674B461
Slave: 4, Output states:
Byte 0: 00000000
Slave: 5, Output data address: 5674B461
Slave: 5, Output states:
Byte 0: 00000000
Slave: 6, Output data address: 5674B461
Slave: 6, Output states:
Byte 0: 00000000
Slave: 2, Index: 0, Value: 1
Initial output value (before setting): 00000000
Val: 00000001
New output value (after setting): 00000001
Slave: 0 has no outputs
Slave: 1 has no outputs
Slave: 2, Output data address: 5674B460
Slave: 2, Output states:
Byte 0: 10000000
Slave: 3, Output data address: 5674B460
Slave: 3, Output states:
Byte 0: 10000000
Slave: 4, Output data address: 5674B461
Slave: 4, Output states:
Byte 0: 00000000
Slave: 5, Output data address: 5674B461
Slave: 5, Output states:
Byte 0: 00000000
Slave: 6, Output data address: 5674B461
Slave: 6, Output states:
Byte 0: 00000000
```

So apparently they share their starting address, as you said above and are combined together. How would is this usually handled by you guys usually? Can i have it such that each slave has their own memory? Or do you simply look at the slave where it starts and access the bit it is shifted to?

nakarlsson, 2023-05-06:
I assume you know of bit masking? Simple as that.
Or use the aligned API that byte align data.

nakarlsson, 2023-11-15:
close due to incativity