OpenEtherCATsociety/SOEM #711: No pdo info in slaveinfo demo

kunijiwa03, 2023-06-06:
Dear all, I am using copley ae2-090-14 driver, and failed to read the pdo info by this cmd: sudo ./slaveinfo eno2 -map. In another way, I configuration the driver by twincat and it can run in csp mode, then reconnect the driver to soem without power off, the pdo info will be printed. Could you please give me some guides?

[slaveinfo without pdo info.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11664312/slaveinfo.without.pdo.info.zip)

nakarlsson, 2023-06-06:
Similar questions have been answered before if you search anong the ”Issues”.

ham-yuan, 2023-06-06:
Hi nakarlsson, thanks for your reply, I searched the topic slaveinfo and copley. For slaveinfo, #655 mentioned need add -map, I added before but didn't work. For copley, I tried #400 but didn't work for red_test. Do you have any other suggestions?

nakarlsson, 2023-06-06:
You can purchase professional support.