OpenEtherCATsociety/SOEM #715: no slaves found error but slaves can be found in TwinCAT

ljin1, 2023-06-19:
Hi,

I am running into an error saying that no slaves are found. However, when I use TwinCAT to search for my devices the slaves can be found. I was wondering why the SOEM library is unable to find the slaves. The device that I have attached to my source PC is a Beckhoff CX 5130 embedded PC, in which there are slaves attached to the embedded PC. 

The below are images of the error and the wireshark log

[image]
[image]

ArthurKetels, 2023-06-20:
It looks like you try to run SOEM under windows and TwinCAT on the same computer. This does not always work because TwinCAT modifies the NIC driver and SOEM will not see any returned packages. Please try on an other computer (either windows or Linux).

ljin1, 2023-06-20:
Ok thanks for the information!

So I tried connecting the devices directly to my source computer and it's
able to identify the slaves.  But now I have another error saying that the
slaves are unable to reach OP state and there's a warning saying that the
Nr of Bytes in input TxPdo are more then transmitted to slaves. What does
this mean and how would I fix this error? Also another thing that I noticed
is that the name of Slave 2 matches the name displayed in TwinCAT, however;
the name of Slave 1 does not match the name displayed in TwinCAT for that
device. Does this have anything to do with the error too?

[image: image.png]

ArthurKetels, 2023-06-21:
You do not provide enough information for me to say anything about your set-up.