OpenEtherCATsociety/SOEM #716: slave account

Edphoenixmax, 2023-06-20:
I wonder how soem detect the number of the slaves.
When I use the slaveinfo.c to get weidmuller's information, it can only get one slave ,that is the coupler,but when I try to get beckhoff's slave,all the slaves'information will show up.
 What I can get from the code is that the number of slave is from wkc=ecx_BRD,and I cannot understand how to get the wkc.
Another question , what is the meaning of the ec_group
looking forward to your answer ,thanks a lot!

ArthurKetels, 2023-06-20:
Only the Weidmuller coupler is an EtherCAT slave. The coupler is a sub ordered master to the module slaves. They run on another (non EtherCAT) protocol. The coupler does however map the module IO in the EtherCAT logical address space. So these modules can be accessed by SOEM via the coupler. 

wkc from ecx_BRD increments by one for each EtherCAT slave in the chain that is able to answer to the requested address space. Simply put if there are 10 slaves then wkc=10.

For ec_group see the tutorial.txt in the doc folder (does nobody read the manual?), and also you can search in this issues section for other posts regarding the subject.

Edphoenixmax, 2023-06-20:
thanks for your answer,I have another question.
At first I think , soem detects the number of slaves by the number of ESC, every slave has one ESC, sothat soem can know the number of slaves. The Weidmuller,is more like a master ,so soem detect the coupler to connect with slaves and the slaves also have ESC,
Is that correct?

ArthurKetels, 2023-06-21:
The Weidmuller coupler acts as a bridge between two different bus systems. Only the coupler has an ESC (EtherCAT Slave Contoller). The modules connected to the coupler have another bus system that is not EtherCAT. Hence they have no ESC and are also not directly visible to SOEM. The coupler thranslates the module configuration into an IO image of the coupler ESC. So more modules will increase the IO size of the coupler ESC. SOEM only needs to see and interact with the coupler.

Edphoenixmax, 2023-06-21:
So this is the difference between Beckhoff and Weidmuller? The Weidmuller modules just increase the IO size,so the number of slave will not change ,just one slave.What about Beckhoff? Why soem can detect the number of modules as slaves?
By the way ,how can I get the  information you talk to me ?
Thanks again for your answer!

closed 2023-06-28 by Edphoenixmax