OpenEtherCATsociety/SOEM #717: Interface eth0 couldn't open with pcap

bereketeshete, 2023-06-21:
I am trying to get slaveinfo on my "TwinCAT-Intel PCI Ethernet Adapter (Gigabit)". I don't how to retrieve this adapter's ifname, so I used the tutorial default `> ./slaveinfo eth0` and SOEM returns with `Interface eth0 couldn't open with pcap` and `No socket connection on eth0`. How can I get slaveinfo on my TwinCAT ethernet adapter?

[image]
[image]

nakarlsson, 2023-06-22:
on windows use the \DEVICE sting instead
Try ./slaveinfo \DEVICE\NPF_{the GUID
}

bereketeshete, 2023-06-22:
Seems like that doesn't solve the issue, SOEM still returns `interface \Device\NPF_ could not open with pcap`

[image]

ArthurKetels, 2023-06-26:
Can you actually get WireShark running on that interface? If not then there is something wrong on the windows driver side.

You could also try the wireless interface. Of course it would not detect any slaves, But it would open the socket, and give you other text output on screen,

mheden, 2023-06-27:
I'll think you need to quote your argument when running from power shell.

bereketeshete, 2023-06-27:
> Can you actually get WireShark running on that interface? If not then there is something wrong on the windows driver side.
> 
> You could also try the wireless interface. Of course it would not detect any slaves, But it would open the socket, and give you other text output on screen,

Thank you, I have currently paused this project. I will try your suggestion when I get back to it.

nakarlsson, 2023-10-25:
question completed

closed 2023-10-25 by nakarlsson