OpenEtherCATsociety/SOEM #718: ec_slave[i] error

Edphoenixmax, 2023-06-26:
After  I configure the IOmap , I change the value of ec_slave[i].outputs to control the digital output and the slaves'lights on and off.
although there are many slaves , when I try to change the number of i to control one specific slave ,there is always 'Syntax error : Unterminated quoted string' occurred.The code can only run successfully when the number of i = 0 ,
So why this error occurred? Is it because I didn't configure IOmap correctly?
Thanks for your answer

ArthurKetels, 2023-06-26:
Please show the line of code where you have the error. This looks like a C language issue and not SOEM.

ArthurKetels, 2023-06-26:
Please have a look at tutorial.txt in the doc folder. There it is explained how to use the IOmap and its pointers.

ec_slave[].outputs is a pointer to a location in IOmap. You can not assign the variable n to the pointer in order to change the IOmap.

closed 2023-06-28 by Edphoenixmax