OpenEtherCATsociety/SOEM #719: ubuntu ros2 humble   cannot open shared object file: No such file or directory

Gates-456, 2023-06-27:
Description of the problem: EtherCAT can work normally under the root user, but it cannot work normally after adding sudo setcap cap_net_admin, cap_net_raw+ep to the compiled target file. It seems that after adding sudo, you cannot link to your own ros2 package lib. How to solve it?
Version: ubuntu ros2 humble
Error:
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [asda_ethercat_node-1]: process started with pid [39742]
[ERROR] [asda_ethercat_node-1]: process has died [pid 39742, exit code 127, cmd '/home/offlines_ws/install/asda_ethercat_driver/lib/asda_ethercat_driver/asda_ethercat_node --ros-args --params-file /tmp/launch_params_ddiqp050 --params-file /tmp/launch_params_jnqxotxy --params-file /tmp/launch_params_p0koe3cm --params-file /tmp/launch_params_hrh4ykd9 --params-file /tmp/launch_params_59v3jony --params-file /tmp/launch_params_z9xrwp97 --params-file /tmp/launch_params_irrrx23b --params-file /tmp/launch_params_iuzb9dey --params-file /tmp/launch_params_1td7anch --params-file /tmp/launch_params_ywl7bwim --params-file /tmp/launch_params_t42err8p --params-file /tmp/launch_params_7zu4mxna'].
[asda_ethercat_node-1] /home/offlines_ws/install/asda_ethercat_driver/lib/asda_ethercat_driver/asda_ethercat_node: error while loading shared libraries: libasda_ethercat_driver_node.so: cannot open shared object file: No such file or directory

Gates-456, 2023-07-11:
seems to add
/etc/ld.so.conf.d$ cat ros2.conf 
/opt/ros/humble/lib/
/etc/ld.so.conf.d$ cat asda_ethercat_driver.conf 
/home/offlines_ws/install/asda_ethercat_driver/lib
/home/offlines_ws/install/msg_type/lib
refer to: https://gitlab.kitware.com/cmake/community/-/wikis/doc/cmake/RPATH-handling

ArthurKetels, 2023-07-11:
An executable with special permissions is excluded from using environment variables. Also those that resolve links to dynamic libraries.  You need to use ldconfig to add the paths to the needed libraries into its cache.

kouge0510, 2025-03-21:
i have met the same problem,and I tried your method,but still not works,still the same error...

kouge0510, 2025-03-21:
I add the rt_ethercat.so in root bashrc to solve  the problem while it shows normal in normal users' environment.