OpenEtherCATsociety/SOEM #722: Beckhoff slaves doesn't store process data

Hugo-Souza, 2023-07-04:
Hello,
I'm a graduate student and I'm doing a project where I need to communicate with my laptop with beckhoff devices (EK1100, EL1809, EL2809 and EL4002).
When executing a test program, which should make a light blink, I manage to initialize the master interface, but when sending the process data to trigger the I/O, instead of staying lit for a while, the light just flashes , giving the impression that the slave was unable to store the data that was sent.

Could someone help me to solve this problem? I've already made some changes to the code, like adding ec_sync but I don't know if it's correct and why this symptom of not "saving" the data in the slave is happening.
Below is the result of the slaveinfo test and the file with the analysis of the packages by wireshark.

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp2s0 succeeded.
Time:1687951497.107 SDO slave:4 index:1c13.01 error:06090011 Subindex does not exist
4 slaves found and configured.
Calculated workcounter 5

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0
PDO mapping according to SII :

Slave:2
 Name:EL1809
 Output size: 0bits
 Input size: 16bits
 State: 4
 Delay: 145[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 07113052 Rev: 00120000
 SM0 A:1000 L:   2 F:00010000 Type:4
 FMMU0 Ls:00000006 Ll:   2 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 TXPDO 0x1A00 Channel 1
     addr b   index: sub bitl data_type    name
  [0x0006.0] 0x6000:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A01 Channel 2
     addr b   index: sub bitl data_type    name
  [0x0006.1] 0x6010:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A02 Channel 3
     addr b   index: sub bitl data_type    name
  [0x0006.2] 0x6020:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A03 Channel 4
     addr b   index: sub bitl data_type    name
  [0x0006.3] 0x6030:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A04 Channel 5
     addr b   index: sub bitl data_type    name
  [0x0006.4] 0x6040:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A05 Channel 6
     addr b   index: sub bitl data_type    name
  [0x0006.5] 0x6050:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A06 Channel 7
     addr b   index: sub bitl data_type    name
  [0x0006.6] 0x6060:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A07 Channel 8
     addr b   index: sub bitl data_type    name
  [0x0006.7] 0x6070:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A08 Channel 9
     addr b   index: sub bitl data_type    name
  [0x0007.0] 0x6080:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A09 Channel 10
     addr b   index: sub bitl data_type    name
  [0x0007.1] 0x6090:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A0A Channel 11
     addr b   index: sub bitl data_type    name
  [0x0007.2] 0x60A0:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A0B Channel 12
     addr b   index: sub bitl data_type    name
  [0x0007.3] 0x60B0:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A0C Channel 13
     addr b   index: sub bitl data_type    name
  [0x0007.4] 0x60C0:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A0D Channel 14
     addr b   index: sub bitl data_type    name
  [0x0007.5] 0x60D0:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A0E Channel 15
     addr b   index: sub bitl data_type    name
  [0x0007.6] 0x60E0:0x01 0x01 BOOLEAN      Input
  SM0 TXPDO 0x1A0F Channel 16
     addr b   index: sub bitl data_type    name
  [0x0007.7] 0x60F0:0x01 0x01 BOOLEAN      Input

Slave:3
 Name:EL2809
 Output size: 16bits
 Input size: 0bits
 State: 4
 Delay: 285[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1003
 Man: 00000002 ID: 0af93052 Rev: 00120000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 SM1 A:0f01 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 140[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1600 Channel 1
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1601 Channel 2
     addr b   index: sub bitl data_type    name
  [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1602 Channel 3
     addr b   index: sub bitl data_type    name
  [0x0000.2] 0x7020:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1603 Channel 4
     addr b   index: sub bitl data_type    name
  [0x0000.3] 0x7030:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1604 Channel 5
     addr b   index: sub bitl data_type    name
  [0x0000.4] 0x7040:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1605 Channel 6
     addr b   index: sub bitl data_type    name
  [0x0000.5] 0x7050:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1606 Channel 7
     addr b   index: sub bitl data_type    name
  [0x0000.6] 0x7060:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1607 Channel 8
     addr b   index: sub bitl data_type    name
  [0x0000.7] 0x7070:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x1608 Channel 9
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x7080:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x1609 Channel 10
     addr b   index: sub bitl data_type    name
  [0x0001.1] 0x7090:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x160A Channel 11
     addr b   index: sub bitl data_type    name
  [0x0001.2] 0x70A0:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x160B Channel 12
     addr b   index: sub bitl data_type    name
  [0x0001.3] 0x70B0:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x160C Channel 13
     addr b   index: sub bitl data_type    name
  [0x0001.4] 0x70C0:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x160D Channel 14
     addr b   index: sub bitl data_type    name
  [0x0001.5] 0x70D0:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x160E Channel 15
     addr b   index: sub bitl data_type    name
  [0x0001.6] 0x70E0:0x01 0x01 BOOLEAN      Output
  SM1 RXPDO 0x160F Channel 16
     addr b   index: sub bitl data_type    name
  [0x0001.7] 0x70F0:0x01 0x01 BOOLEAN      Output

Slave:4
 Name:EL4002
 Output size: 32bits
 Input size: 0bits
 State: 4
 Delay: 440[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1004
 Man: 00000002 ID: 0fa23052 Rev: 00140000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   4 F:00010024 Type:3
 SM3 A:1180 L:   0 F:00000000 Type:4
 FMMU0 Ls:00000002 Ll:   4 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 03 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 140[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0002.0] 0x7000:0x01 0x10 INTEGER16    Analog output
  [0x0004.0] 0x7010:0x01 0x10 INTEGER16    Analog output
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program
``` 
Wireshark=[packet_lamp_blink_sync.pcap.gz](https://github.com/OpenEtherCATsociety/SOEM/files/11950342/packet_lamp_blink_sync.pcap.gz)

My code=[lampblinksync.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11950358/lampblinksync.zip)

ArthurKetels, 2023-07-04:
When slaves are in safe-OP or OP state the PDO transfer needs to be in regular intervals. In your Wireshark capture the interval is between a few us to many seconds (up to 67s). All output slaves have a watchdog timer that sets the slave state to safe-OP+error and disables all outputs after approx. 100ms. And this is exactly what happened in your wireshark capture.

For how to build a SOEM application please read the tutorial.txt and have a look at the red_test example.

Hugo-Souza, 2023-07-26:
Thank you very much for the tips Arthur, and I apologize for the delay in returning an answer.

Following yours, I redid my test program and it worked, assigning the logical values ​​correctly in the modules of the digital output device EL2809.

Continuing my sequence of tests, I was also able to read the logical values ​​of a button through the EL1809 digital input device, but during the execution of this test, the handler error thread of the slaves in the operational state sometimes shows that the slaves have returned to the operational state, but analyzing the wireshark capture, no slave left the operational state.

[image]

I have two questions:
1. Is this  handle slave thread symptom normal? Appear in the program and the capture does not show the change of state?
2. How can I calculate Cycle Time? I'm using the suggested value in red_test, but I don't think it would be ideal for my network

Below is the input device test code and the capture performed by wireshark:
[InputTestCode](https://github.com/OpenEtherCATsociety/SOEM/files/12176488/buttonpresstest.zip)
[WiresharkCapture](https://github.com/OpenEtherCATsociety/SOEM/files/12176489/packets.pcapng.gz)

closed 2024-10-16 by Hugo-Souza