OpenEtherCATsociety/SOEM #725: Can not change all slaves' al status from safe-op to OP successfully every time

TDonkey, 2023-07-11:
Hi, guys, l use SOEM for Ethercat slaves and it always works fine;

oneday , l try to communicate with a slave from a new brand, sometimes it can transfer from safe-op to op sucessfully,but sometimes certain slaves fail to get OP with AL statusCode Error=0x001b : sync manager watchdog. 
l patch the wireshark info, slave info and intial code here.
 
when l use SOEM, l usually come here and learn a lot from you guys' patiently explain which always helps me to solve problems;
Any advices will be helpful, thanks! 

[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/12015969/wireshark.zip)
[slave_info.txt](https://github.com/OpenEtherCATsociety/SOEM/files/12015976/slave_info.txt)
[initial_code.txt](https://github.com/OpenEtherCATsociety/SOEM/files/12016002/initial_code.txt)

TDonkey, 2023-09-01:
Problem solved

closed 2023-09-01 by TDonkey

ArthurKetels, 2023-09-01:
WOuld be nice if you can share HOW you solved it. Perhaps it prevents others from making the same mistake?

TDonkey, 2023-09-06:
l find the slave from new brand has a more strict sync manager watchdog setup, which leads to unstable communication results that sometimes l can't transfer from safe-op to op successfully.

So, l improve my code for communication initialization in steps below:
1.create thread for cyclical ECAT frames send/receive (just created ,not trigged)
2.ec_init()
3.ec_config_init()
4.ec_slave[i + 1].PO2SOconfig = pdoSetupIsmcServo; //setup pdo settings func for your slave
5.ec_configdc()
6.ec_dcsync0(i + 1, true, ctime * 1000, ctime * 500)
7.ec_config_overlap_map(IOMap);
8.trigged your thread created in step 1 for ECAT frames send/receive obeying your DC period (key point to slove my problems!!!)
9.request tranfer to SAFE-OP
10.request tranfer to OP

l wish these infos will help others

ArthurKetels, 2023-09-06:
Thanks for the explanation! This is useful for other too. 

It is true that more slaves require stable synchronous communication in safe-OP. Sometimes this is really necessary, but more often it is just lazy programming.