OpenEtherCATsociety/SOEM #726: Some questions about the ec_sync function in red_test.c

brentonchris, 2023-07-13:
/* PI calculation to get linux time synced to DC time */
void ec_sync(int64 reftime, int64 cycletime , int64 *offsettime)
{
   static int64 integral = 0;
   int64 delta;
   /* set linux sync point 50us later than DC sync, just as example */
   delta = (reftime - 50000) % cycletime;
   if(delta> (cycletime / 2)) { delta= delta - cycletime; }
   if(delta>0){ integral++; }
   if(delta<0){ integral--; }
   *offsettime = -(delta / 100) - (integral / 20);
}
....
 if (ec_slave[0].hasdc)
 {
            /* calulate toff to get linux time and DC synced */
            ec_sync(ec_DCtime, cycletime, &toff);
  }

**I saw ec_ DCtime is extracted from the data frame when receiving pdo data, and is the following code:**
memcpy(&le_DCtime, &(rxbuf[idx][idxstack->dcoffset[pos]]), sizeof(le_DCtime));
 *(context->DCtime) = etohll(le_DCtime);

**I don't know What time is ec_DCtime? 
Which slave station is it from?**

nakarlsson, 2023-07-13:
It is the timestamp on frame arrival at first DC able slave. Eg DC ref slave.

brentonchris, 2023-08-10:
Thank you for your reply. This issue has been clarified.

closed 2023-08-10 by brentonchris