OpenEtherCATsociety/SOEM #727: Empty value or zero when casting

jeanpaulsq, 2023-07-19:
Hello everyone,

After maping the IOMap and bind the PDO, I want to associate the inputs and outputs with 2 structures (explained later).
```
uint8_t* in = ec_slave[slaveID].inputs;
uint8_t* out = ec_slave[slaveID].outputs;
```

If I try to execute directly this pointers it does not print anything.
`
logger_->critical("Out:{} In:{}, Max", out,in);
`

_"... [Thread:120463] [critical] : Out: In:, Max"_

But if I cast them into a pointer then I can print the value:
```
logger_->info("Out:{},{} In:{},{}", (uintptr_t)ec_slave[slaveID].outputs, ec_slave[slaveID].Obytes,
            (uintptr_t)ec_slave[slaveID].inputs, ec_slave[slaveID].Ibytes
```

_"...[Thread:120463] [info] : Out:94292108689600,1 In:94292108689601,77"_

I want to transform the pointers into two structures:

```
struct TxPdo {
    uint8_t status;
    uint32_t warningsErrorsFatals;
    float forceX;
    float forceY;
    float forceZ;
    float torqueX;
    float torqueY;
    float torqueZ;
    uint16_t forceTorqueSaturated;
    float accelerationX;
    float accelerationY;
    float accelerationZ;
    uint8_t accelerationSaturated;
    float angularRateX;
    float angularRateY;
    float angularRateZ;
    uint8_t angularRateSaturated;
    float temperature;
    float estimatedOrientationX;
    float estimatedOrientationY;
    float estimatedOrientationZ;
    float estimatedOrientationW;
} __attribute__((packed)); // To avoid data alignment

struct RxPdo {
    uint8_t digitalOutput = 0;
} __attribute__((packed));
```

I've tried with 2 different ways:
- First way:
```
TxPdo* txpdo_;
txpdo_ = reinterpret_cast<TxPdo*>(in);
printf("Force: %f, %f %f\n", txpdo_->forceX, txpdo_->forceY, txpdo_->forceZ);

RxPdo* rxpdo_;
rxpdo_ = reinterpret_cast<RxPdo*>(out);
printf("DigitalOutput: %u\n", rxpdo_->digitalOutput);
```

- Second way:
```
TxPdo txpdo_;
memcpy(&txpdo_, in, sizeof(TxPdo));
printf("Force: %f, %f %f\n", txpdo_.forceX, txpdo_.forceY, txpdo_.forceZ);

RxPdo rxpdo_;
memcpy(&rxpdo_, out, sizeof(RxPdo));
printf("DigitalOutput: %u\n", rxpdo_.digitalOutput);
```

But the result is the same:
_Force: 0.000000, 0.000000 0.000000
DigitalOutput: 0_

I've seen another issue related to this [#423](https://github.com/OpenEtherCATsociety/SOEM/issues/423), but they solved that modifying the file _ethercatcoe.c_ directly from the library, and I don't have access to this file in the project. In fact, I just see the *.a file.

Thank you in advance.

ArthurKetels, 2023-07-19:
Can you post the output from slaveinfo -map? Then it is obvious if we have a slave with a firmware issue or not.

jeanpaulsq, 2023-07-19:
I don't know how to do it, could you explain me please?

easters14, 2023-12-01:
> I don't know how to do it, could you explain me please?

Just build the test by following README.md, and run slaveinfo binary file under the test directory