OpenEtherCATsociety/SOEM #731: Invalid output configuration

innovatitw, 2023-08-07:
I am using Raspberry Pi 4 to develop ETHERCAT SOEM on Ubuntu, and working fine with peripheral devices such as Mitsubishi, Sanyo, Omron, and Delta IO module. 
However, when I use DELTA EC8124 ADC module, the slaveinfo command shows an error message of "invalid output configuration". What is the cause of this error and how to solve it. Please help, thanks a lot.

slaveinfo output
https://app.box.com/s/l7u0fp1kesvwwntsl4a7x28dpd2p5rla
https://app.box.com/s/w5adjuoe18xhlabzo7wdtzc9z1tz0ky4

innovatitw, 2023-08-17:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth0 succeeded.
2 slaves found and configured.
Calculated workcounter 1
Not all slaves reached safe operational state.
Slave 2 State=12 StatusCode=  1d : Invalid output configuration

Slave:1
 Name:R1-EC5500
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 000001dd ID: 00005500 Rev: 00100000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0

Slave:2
 Name:R1-EC8124
 Output size: 0bits
 Input size: 64bits
 State: 18
 Delay: 155[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: 000001dd ID: 00008124 Rev: 00100001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   8 F:00010024 Type:3
 SM3 A:11c0 L:   8 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:11c0 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 03 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 300[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

innovatitw, 2023-08-17:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth0 succeeded.
2 slaves found and configured.
Calculated workcounter 1
Not all slaves reached safe operational state.
Slave 2 State=12 StatusCode=  1d : Invalid output configuration

Slave:1
 Name:R1-EC5500
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 000001dd ID: 00005500 Rev: 00100000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0

Slave:2
 Name:R1-EC8124
 Output size: 0bits
 Input size: 64bits
 State: 18
 Delay: 155[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: 000001dd ID: 00008124 Rev: 00100001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   8 F:00010024 Type:3
 SM3 A:11c0 L:   8 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:11c0 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 03 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 300[mA]
 only LRD/LWR:0
 CoE Object Description found, 21 entries.
0x1000      "Device type"                                 [VAR]
    0x00      "Device type"                                 [UNSIGNED32       R_R_R_]      0x00040191 / 262545
0x1001      "Error register"                              [VAR]
    0x00      "Error register"                              [UNSIGNED8        R_R_R_]      0x00 / 0
0x1008      "Device name"                                 [VAR]
    0x00      "Device name"                                 [VISIBLE_STR(80)  R_R_R_]      "R1-EC8124"
0x1009      "Hardware version"                            [VAR]
    0x00      "Hardware version"                            [VISIBLE_STR(32)  R_R_R_]      "n.a."
0x100a      "Software version"                            [VAR]
    0x00      "Software version"                            [VISIBLE_STR(32)  R_R_R_]      "1.10"
0x1018      "Identity"                                    [RECORD  maxsub(0x04 / 4)]
    0x00      "SubIndex 000"                                [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "Vendor ID"                                   [UNSIGNED32       R_R_R_]      0x000001dd / 477
    0x02      "Product code"                                [UNSIGNED32       R_R_R_]      0x00008124 / 33060
    0x03      "Revision"                                    [UNSIGNED32       R_R_R_]      0x00100001 / 1048577
    0x04      "Serial number"                               [UNSIGNED32       R_R_R_]      0x00000000 / 0
0x10f1      "Error Settings"                              [RECORD  maxsub(0x02 / 2)]
    0x00      "SubIndex 000"                                [UNSIGNED8        R_R_R_]      0x02 / 2
    0x01      "Local Error Reaction"                        [UNSIGNED32       RWRWRW]      0x00000001 / 1
    0x02      "Sync Error Counter Limit"                    [UNSIGNED16       RWRWRW]      0x0004 / 4
0x1601      "RxPDO-Map"                                   [RECORD  maxsub(0x01 / 1)]
    0x00      "SubIndex 000"                                [UNSIGNED8        R_R_R_]      0x00 / 0
[... 44 lines cut ...]
    0x00      "OS Ratio select"                             [UNSIGNED16       RWRWRW]      0x0000 / 0
0x2002      "Enable Input"                                [RECORD  maxsub(0x04 / 4)]
    0x00      "SubIndex 000"                                [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "Ch0 Enable"                                  [UNSIGNED16       RWRWRW]      0x0000 / 0
    0x02      "Ch1 Enable"                                  [UNSIGNED16       RWRWRW]      0x0000 / 0
    0x03      "Ch2 Enable"                                  [UNSIGNED16       RWRWRW]      0x0000 / 0
    0x04      "Ch3 Enable"                                  [UNSIGNED16       RWRWRW]      0x0000 / 0
0x2003      "Average select"                              [VAR]
    0x00      "Average select"                              [UNSIGNED32       RWRWRW]      0x00000001 / 1
0x20fe      "Trim Value select"                           [RECORD  maxsub(0x04 / 4)]
    0x00      "SubIndex 000"                                [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "Trim Value select ch0"                       [UNSIGNED16       RWRWRW]      0x2710 / 10000
    0x02      "Trim Value select ch1"                       [UNSIGNED16       RWRWRW]      0x2710 / 10000
    0x03      "Trim Value select ch2"                       [UNSIGNED16       RWRWRW]      0x2710 / 10000
    0x04      "Trim Value select ch3"                       [UNSIGNED16       RWRWRW]      0x2710 / 10000
0x20ff      "Trim Value select"                           [VAR]
    0x00      "Trim Value select"                           [UNSIGNED8        RWRWRW]      0xff / 255
0x6401      "AO inputs channel"                           [RECORD  maxsub(0x04 / 4)]
    0x00      "SubIndex 000"                                [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "Analog input ch0"                            [UNSIGNED16       R_R_R_]      0x8000 / 32768
    0x02      "Analog input ch1"                            [UNSIGNED16       R_R_R_]      0x8000 / 32768
    0x03      "Analog input ch2"                            [UNSIGNED16       R_R_R_]      0x8000 / 32768
    0x04      "Analog input ch3"                            [UNSIGNED16       R_R_R_]      0x8000 / 32768
End slaveinfo, close socket
End program

IceTea2, 2023-11-02:
Hey, were you able to figure this one out? 
I have the same problem it seems..

closed 2024-03-05 by innovatitw