OpenEtherCATsociety/SOEM #732: Porting SOEM Code on RTOS

brentonchris, 2023-08-10:
Has the SOEM code been ported on RTOS?

nakarlsson, 2023-08-10:
All non *nix and Win32 targets in HW/AL layers are RTOSs

brentonchris, 2023-08-10:
I want to port the SOEM code on the ThreadX RTOS.
Are there any successful cases of transplantation that can be referred to?

nakarlsson, 2023-08-10:
I don't know any.

awolf-illumina, 2023-08-21:
@brentonchris Any luck? I'm looking to do a port for ThreadX as well

brentonchris, 2023-08-22:
> @brentonchris Any luck? I'm looking to do a port for ThreadX as well

We already run SOEM without an operating system, but now we want to port it to threadx. However, there is a large error between the time of sending data and the target time point, it is the ec_sync in the SOEM code.

WillyTuring, 2023-09-01:
> 

Hi chris, "However, there is a large error between the time of sending data and the target time point" what is this problem exaclty? I am considering migrating from Linux OS to MCU or possibly combining with FPGA as well for my application here.

nakarlsson, 2023-10-25:
@brentonchris , can we close this issue?

closed 2023-11-15 by nakarlsson

brentonchris, 2023-11-16:
> Guess we can close it

OK,close it.