OpenEtherCATsociety/SOEM #734: Is there any possible way to use the library with FPGA?

WillyTuring, 2023-09-01:
Hi folks,

The title is basically my question. I am controlling a 6 dof robotic arm with torque control mode, i have built a dynamic model for the whole robot, the mathematical calculation is quite intense. I want to make it in parallel to speed up the calculation process, then reduce the Master-Slave communication cycle from 4ms to 1ms.

The problem with MCU is that it runs in series and the calculation speed couldn't be sped up so the real-time communication between master and slave might get jeopardized.

ArthurKetels, 2023-09-01:
Professionally I run high DOF (8+) torque controlled forward and inverse kinematics on a standard MCU at 1kHz without any issues. Just be smart with your algorithm. 

Adapting SOEM to run on an FPGA as VHDL is al tall order. Much better to use a soft-CPU on the FPGA. Or a modern FPGA with integrated hard-CPU. Many projects with SOEM have been done like this.

WillyTuring, 2023-09-01:
Hi Arthur,

You are back.

1.Yes, i am planning to use a board with zynq 7000 series chip, what do you say about the OS, should i run embedded linux + xenomai / Preempt-RT or freeRTOS, why?

2.You mentioned about the inverse dynamic model, I think mine is fairly good now, but i am still struggling about the feedback control part, see the command sent to the driver is the torque command, and i am using the velocity error the as the feedback control signal with PI controller.
[image]
[image]
Take a look at those two images, those are the velocity curve with unit rad/s for joint 1 and joint 2, orange one is the desired one, blue one is the actual one, to be honest, i don't know how to tune the PI controller rigorously in a systematic way especially the gains of one joint will also affect the other joint. What do you say how should i tackle this feedback control loop, subjectively i feel like this primitive PI controller is not enough for the best performance.

WillyTuring, 2023-09-07:
Hi Arthur,

I did use the velocity error as the feedback signal, but the correction is still for acceleration, for example if the actual velocity is smaller than the desired ones, we will increase the acceleration.

We can not use the acceleration error as the feedback signal since there is a discrepancy already between the theoretical modeling and the actual system, meaning that due to the hard-to-model factor like the friction and the alike, even if the actual acceleration is the same as the goal acceleration, the actual velocity won't be the same as the desired velocity.

ArthurKetels, 2023-09-09:
> if the actual velocity is smaller than the desired ones, we will increase the acceleration

What you tell here is the definition of a second order control loop. So it will behave as a second order system. Optimal feedback control is a cascade of first order control loops. In your case, acceleration error controls acceleration, velocity error controls velocity, position error control position. The statement that it is inconvenient or hard does not change facts.

All commercial robotic control arm controls that I know of use feed forward in combination with feed back. The most important feed forward modelling is, gravity, inertia and friction. Feed back control is often only velocity and position. The best ones also have force feedback control. 

Because a robotic arm is a series actuator, every joint action also gives a response on all other joints. These kinematic effects are always propagated via the feed forward controller to all joints. In academic literature this is referred to as the Jacobian and the Hessian transform. You can not hope to get good and accurate trajectory control without those.

Sorry for this very off topic lecture about control theory in a forum about a communication library. So I would like to keep it at that.

WillyTuring, 2023-09-09:
Ok, we will stop there in here then.

One last note the feedforward process is done via my 6 dof dynamic model, this model takes the mass, link length and effective inertia into account realtime in every control cycle.

One last question, do you happen to know any groups about the control theory where could have much deeper discussions about this topic? It would be really helpful if you could share one or two.

closed 2023-09-27 by WillyTuring