OpenEtherCATsociety/SOEM #736: this is the initcmd of issue 735

Edphoenixmax, 2023-09-01:
<InitCmd>
                    <Transition>IP</Transition>
                    <Transition>IB</Transition>
                    <Comment>set physical address</Comment>
                    <Requires>cycle</Requires>
                    <Cmd>2</Cmd>
                    <Adp>0</Adp>
                    <Ado>16</Ado>
                    <Data>E903</Data>
                    <Cnt>1</Cnt>
                    <Retries>3</Retries>
                </InitCmd>

ArthurKetels, 2023-09-01:
OK, and what is the code that you wrote to replicate this command?

Edphoenixmax, 2023-09-02:
I just write a funtion to get these initcmds,according to the Cmd to choose proper function.
Like this, I use ecx_APWR() to write this data in the Ado,If I donnot use this command , the Coe mailbox initcmds can be wrote through sdowrite function , but other slave initcmds like set DC activation especially the transition is PS,cannot be write properly.
So I just cannot figure it out ,what is the conflict between Coe and this initcmd.
Thanks again for your reply!

ArthurKetels, 2023-09-06:
I think I understand what your problem actually is. 
From the initcmd: `<Comment>set physical address</Comment>`

This will reprogram the address assigned to the slave. If SOEM had another address before it will no longer be possible to communicate with the slave on the original address. SOEM still has the original address stored in its data structure.

Two options, skip this command, or update the SOEM data with the address from the INI file (in this case 0x03e9).

In general you either configure the slaves with only the INI file, or only the SOEM auto detect. Not a mix between the two.

Edphoenixmax, 2023-09-06:
Thanks for your answer, you are the best!
But I think the address SOEM auto detect is same as the initcmd, why it still cannot go through?
If I skip this command, ther will be  conflicts of  other initcmds like:
<InitCmd>
                    <Transition>IP</Transition>
                    <Transition>SP</Transition>
                    <Transition>SI</Transition>
                    <Transition>OP</Transition>
                    <Transition>OI</Transition>
                    <Comment>clear sm 2 (outputs)</Comment>
                    <Requires>cycle</Requires>
                    <Cmd>5</Cmd>
                    <Adp>1001</Adp>
                    <Ado>2064</Ado>
                    <Data>0000000000000000</Data>
                    <Retries>3</Retries>
                </InitCmd>
                <InitCmd>
                    <Transition>PS</Transition>
                    <Comment>set sm 2 (outputs)</Comment>
                    <Cmd>5</Cmd>
                    <Adp>1001</Adp>
                    <Ado>2064</Ado>
                    <Data>00180C0064000100</Data>
                    <Cnt>1</Cnt>
                    <Retries>3</Retries>
                </InitCmd>
and so on
Will it cause other influence of other functions of SOEM like the configaddr if I choose to configure the slaves only the INI file?
Should I change the code of ecx_config_init or abandon the whole ecx_config_init and just configure the slaves by INI file？

Edphoenixmax, 2023-09-06:
<InitCmd>
                    <Transition>IP</Transition>
                    <Transition>SP</Transition>
                    <Transition>SI</Transition>
                    <Transition>OP</Transition>
                    <Transition>OI</Transition>
                    <Comment>clear sm 2 (outputs)</Comment>
                    <Requires>cycle</Requires>
                    <Cmd>5</Cmd>
                    <Adp>1001</Adp>
                    <Ado>2064</Ado>
                    <Data>0000000000000000</Data>
                    <Retries>3</Retries>
                </InitCmd>
                <InitCmd>
                    <Transition>PS</Transition>
                    <Comment>set sm 2 (outputs)</Comment>
                    <Cmd>5</Cmd>
                    <Adp>1001</Adp>
                    <Ado>2064</Ado>
                    <Data>00180C0064000100</Data>
                    <Cnt>1</Cnt>
                    <Retries>3</Retries>
                </InitCmd>

ArthurKetels, 2023-09-06:
As mentioned above, do not mix. Configure by INI file. There is one big BUT : you also need to change the PDO transfer, because that will also be ruled by the INI file. In essence by using the INI file you remove any intelligence from the SOEM code.

closed 2023-10-25 by Edphoenixmax