OpenEtherCATsociety/SOEM #738: A frame with multi-nodes info

Lambo0604, 2023-09-05:
I have a question about redundancy about SOEM-1.4.0.  Do we support to send and recv a frame with multi-nodes info, such as below:
[image]
I think  it can save the system’s performance in frequently sending and receiving messages.
Do we support to send and recv this kind of frames?

ArthurKetels, 2023-09-06:
Yes and no, There is a low level function to add datagrams to an existing frame. It is used in the PDO data transfer.

So you could build a frame that looks like the above example. But you need to know what you are doing. I have been thinking of making functions to do this in a user friendly way. But I always ended up with a messy API. 

But for this specific use case (reading the individual slave states) it is very well possible to make a special function. Corner case is when there are more slaves than fit in a frame. Also slaves can be member of different groups (so addresses are no longer continuous).

Most often I do not find a need for these combined datagrams. It is often better to spread individual status reads over multiple cycles. There is seldom a need to have this data at high temporal resolution. You know right away from the WKC if transfers are successful or not. Figuring out which slave is the cause can wait  a bit longer. 

```
/** Add EtherCAT datagram to a standard ethernet frame with existing datagram(s).
 *
 * @param[in] port        = port context struct
 * @param[out] frame      = framebuffer
 * @param[in]  com        = command
 * @param[in]  idx        = index used for TX and RX buffers
 * @param[in]  more       = TRUE if still more datagrams to follow
 * @param[in]  ADP        = Address Position
 * @param[in]  ADO        = Address Offset
 * @param[in]  length     = length of datagram excluding EtherCAT header
 * @param[in]  data       = databuffer to be copied in datagram
 * @return Offset to data in rx frame, usefull to retrieve data after RX.
 */
uint16 ecx_adddatagram(ecx_portt *port, void *frame, uint8 com, uint8 idx, boolean more, uint16 ADP, uint16 ADO, uint16 length, void *data)
```

Lambo0604, 2023-09-15:
Thank u very much! I will try this function in my project. 
Thanks again!

nakarlsson, 2023-10-25:
Issue seems completed

closed 2023-10-25 by nakarlsson