OpenEtherCATsociety/SOEM #740: Larger EC_MAXGROUP is needed (defaults to 2)

MasterYip, 2023-09-14:
I've been working on wrapping an EthercatBusMgr based on SOEM. I noticed the EC_MAXGROUP(ethercatmain.h) is limited to 2 by default, which has brought me some problems when I try to set up multiple groups.

I wonder if it is possible to change this default? Or is there any reason for that?

ArthurKetels, 2023-09-14:
It is a simple reason. Almost nobody uses groups, and it will cause memory overhead for embedded systems (there are lots of them). If you want to use it, then simply increase the default.

MasterYip, 2023-09-15:
thanks a lot~

MasterYip, 2023-09-15:
Is the embedded system you mentioned `embedded linux`? Can SOEM run on any other kind of mono-chip microcomputer(e.g. stm32)?

ArthurKetels, 2023-09-15:
SOEM was developed to have a minimal and fixed memory footprint (no dynamic memory allocator needed). SOEM does not assume an OS is available and can run bare-metal. It even has been ported to 8 bit microcontrollers with 128kb flash memory. A STM32 is no problem at all and delivers quite good performance (>10kHz cycle rate).

MasterYip, 2023-09-15:
I See. Appreciate it again for your reply:)

closed 2023-09-15 by MasterYip