OpenEtherCATsociety/SOEM #741: SAFE_OP + ERROR after sending SDO

open04, 2023-09-23:
Hi, first of all, thank you for making this library.

As the title stated, Im having SAFE_OP + ERROR after sending SDO. I modified the simple_test to try to rotate the motor.
Here is the modified simple_test:

`/** \file
 * \brief Example code for Simple Open EtherCAT master
 *
 * Usage : simple_test [ifname1]
 * ifname is NIC interface, f.e. eth0
 *
 * This is a minimal test.
 *
 * (c)Arthur Ketels 2010 - 2011
 */

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

typedef struct PACKED
{
  int8_t modes_of_operation;
  uint16_t control_word;
  int32_t _position;
  int32_t _velocity;
  int16_t _torque;
} output_pdo_t;

typedef struct PACKED
{
  int8_t modes_of_operation_display;
  uint16_t status_word;
  int32_t position_actual;
  int32_t velocity_actual;
  int16_t torque_actual;
} input_pdo_t;

input_pdo_t*input_pdo_;
output_pdo_t*output_pdo_;

void simpletest(char *ifname)
{
    int oloop, iloop, chk;
    needlf = FALSE;
    inOP = FALSE;

   printf("Starting simple test\n");

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);
[... 192 lines cut ...]
    }
}

int main(int argc, char *argv[])
{
   printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");

   if (argc > 1)
   {
      /* create thread to handle slave error handling in OP */
//      pthread_create( &thread1, NULL, (void *) &ecatcheck, (void*) &ctime);
      osal_thread_create(&thread1, 128000, &ecatcheck, (void*) &ctime);
      /* start cyclic part */
      simpletest(argv[1]);
   }
   else
   {
      printf("Usage: simple_test ifname1\nifname = eth0 for example\n");
   }

   printf("End program\n");
   return (0);
}`

there is no problem when I tried to run the original simple_test script. am I missing something?

open04, 2023-09-23:
Here is the log:

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 36 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
SDO write
ERROR : slave 1 is in SAFE_OP + ERROR, attempting ack.
WARNING : slave 1 is in SAFE_OP, change to OPERATIONAL.
MOO : 0
CW128
ERROR : slave 1 is in SAFE_OP + ERROR, attempting ack.
WARNING : slave 1 is in SAFE_OP, change to OPERATIONAL.
CW6
CW7
CW15

ArthurKetels, 2023-09-25:
Yes the behaviour makes perfect sense. When a slave goes to operational a watchdog timer is started on the slave that monitors cyclic PDO transfers. After more than 100ms of no transfers the watchdog kicks in and the slave goes in safe-OP+error state.

You need to have a separate task that does continuous process-data transfers. See many posts here on how to do that (use the search function).

open04, 2023-09-26:
Hi Arthur, Thanks! Searched for the same issue here and I can now control my motor. Thanks again

closed 2023-09-26 by open04