OpenEtherCATsociety/SOEM #742: Incorrect parent of the last slave

Cdfghglz, 2023-09-26:
With some newer hardware (EPOS Disk 2022) I am having troubles initializing the bus properly. The same code workes with older hw (EPOS Compact). EtherCAT version on both according to datasheets is the same.

With both hw versions there are 12 slaves. With the new ones during initialization, the last slave reports id 1 as parent instead of 11. The same happens for n_slaves=11, but for numbers up to 5 all seems fine.

The topology in `ethercatconfig.c` seems a little suspicious to me despite I did not dig too deep yet.

```bash
...
Slave 6 topology: 23095
Slave 7 topology: 23095
Slave 8 topology: 23095
Slave 9 topology: 23095
Slave 10 topology: 22039
Slave 11 topology: 23095
```

Version of soem 1.3.3.

Does anybody have an idea whether this is an issue of the controller firmware or has to do with the master implementation?

Cdfghglz, 2023-09-27:
Was a swapped connector :see_no_evil: . We checked like 100x and were confident cause the datasheet says hardware damage if in/out are swapped... I am surprised it worked so well :-D it read the positions like charm.

Thanks!

ArthurKetels, 2023-09-27:
There is no damage when swapping IN OUT ports. That is ridiculous. 

As you know all EtherCAT topology is functionally just a circle. Even if it physically looks different. So when you swap in and out ports it means the ESC is somewhere else in the circle from master through all slaves and back again. SOEM will tell you the effective slave order in the circle.

Cdfghglz, 2023-09-28:
Thank you for the info, closing.

closed 2023-09-28 by Cdfghglz