OpenEtherCATsociety/SOEM #744: information get from 0x1c12

Edphoenixmax, 2023-09-28:
I want to ask soem how to get the information of 0x1c12 and 0x1c13.
Are these information stored in the ESC of slaves?
If I use the slaveinfo -sdo function but I cannot get the information of  0x1c12 and 0x1c13, does it mean there is no information in ESC?
Thanks for your reply!

ArthurKetels, 2023-09-28:
I think your question is about SDO 0x1c12. SDO is a parameter exchange protocol that runs over EtherCAT mailbox protocol. 

Not all slaves have this implemented. You can check this with slaveinfo CoE capabilities. If there is no CoE or even mailbox in a slave then the SDO objects are also not available. 

It is possible to store an equivalent in the slave EEPROM in the SII. SOEM can read this out too. But again, not all slaves support this.

nakarlsson, 2023-10-25:
@Edphoenixmax , it is good manners to close questions on completion.

closed 2023-10-25 by nakarlsson