OpenEtherCATsociety/SOEM #745: multiple master with SOEM

paul218, 2023-09-29:
Hello. I am trying to use two different networks: the first is to get data from an EtherCAT Device, and the second is to control a motor. I created a class to use SOEM and instance it. 
The first device Is connected to a USB-Ethernet Adapter and the other to a native Ethernet port. 
I tested my class with one device and different ports. When I used it one by one everything worked well, but when I joined the two SOEM classes one slave lost the communication. 

Is possible to have two SOEM masters in the same program?

ArthurKetels, 2023-10-01:
Yes, as long you have to separate contexts. 

You can not use any of the EC_VER1 API as it does not have variable context. So not ec_config() but ecx_config().

It is best to define EC_VER2 in your program, this way the old API is no longer available.

paul218, 2023-10-02:
Thanks a lot!
Are there some examples using the EC_VER2?

nakarlsson, 2023-10-25:
Basic sample for intime, but you'll get the idea.

https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/intime/ec_master/ec_master.c

closed 2023-10-25 by nakarlsson