OpenEtherCATsociety/SOEM #747: Any suggestions for optimizing NIC drivers？

casinia, 2023-10-07:
We use the red_test.c from the SOEM test/linux directory with a cycle 1kHz and modified it to measure some timing behaviour.And i know we must optimize the NIC driver(Intel I210) for real-time latency, but we‘re at a loss. Can you give us some suggestions？

nakarlsson, 2023-10-11:
Make sure "Interrupt throttling" is disabled.

nakarlsson, 2023-10-25:
@casinia , can we close this issue?

casinia, 2023-10-26:
yes，thanks------------------ Original ------------------From: nakarlsson  ***@***.***>Date: Wed,Oct 25,2023 6:26 PMTo: OpenEtherCATsociety/SOEM ***@***.***>Cc: casinia ***@***.***>, Mention ***@***.***>Subject: Re: [OpenEtherCATsociety/SOEM] Any suggestions for optimizing NIC drivers？ (Issue #747)
@casinia , can we close this issue?

[
{
"target": "https://github.com/OpenEtherCATsociety/SOEM/issues/747#issuecomment-1778964139",
"url": "https://github.com/OpenEtherCATsociety/SOEM/issues/747#issuecomment-1778964139",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2023-10-26 by nakarlsson