OpenEtherCATsociety/SOEM #749: use ec_sync0 error 0x002d

Edphoenixmax, 2023-10-12:
I met a error 0x002d--no SYNC0 or SYNC1 received when I use ecx_dcsync0 function, and it caused the slave cannot change state from safeop to op.
I use wireshark to catch frames pcap and I find the value in the register 0x09a0 is correct at first time,and then it send FPWR to write 0xffffff to this register, and I have no idea what happens.
I am sorry I cannot upload the pcap to u, 
Do you know what is the problem is or can you teach me some ways to figure it out or get some specific information?
Thank you a lot!

ArthurKetels, 2023-10-12:
Sorry, not enough information to give a sensible answer. Did you first enable DC with ecx_configdc()?
For the rest you could search in the issues posts, more than enough examples on how to do this.

Edphoenixmax, 2023-10-12:
thanks 
Another question, when I use codesys to config this sync0time, then I use slaveinfo -sdo and I find in 0x1c32 0x02 is cycletime that you set.
But if I just use configdc and ecx_sync0 , the cycletime will not be set, and I donnot know why,
So ,can the ecx_sync0 function set repetitive sync signal to slaves?

Edphoenixmax, 2023-10-12:
the sync cycletime is 0x1c32 and 0x1c33 0x0a , I shoule correct this 
Thanks a lot !

closed 2023-10-25 by Edphoenixmax