OpenEtherCATsociety/SOEM #751: How to confirm that DC has been settled correctly

Edphoenixmax, 2023-10-16:
I have three problems:
1.If I have two slaves, how to make sure that I have settled the DC correctly? Can I get it from the pcap I caught by wireshark?
2.I use codesys and TwinCAT to settle DC, and I export the ENI file , there is some differences confuse me. There is a Initcmd called Set DC activation , the value of this cmd is 003( I understand that is sync0 activated ),but the address of this cmd is not 0x981 but 0x980,and I   check the pcap caught by wireshark , I find that the value of ox981 is in the frame of 0x980, and I donnot know why.(the two picture is ecat.ado==0x980 filter and ecat==0x981 filter)
3.I use ecx_sync0 function and ENI file to set cycletime after configdc(),but I check the frame from wireshark, the last value of 0x981 is 0 instead of 3 , and I cannot change to state OP.
Thank you very much!
  
[image]
[image]

ArthurKetels, 2023-10-17:
1. Register System Time Difference (0x092C:0x092F), read this value cyclically and plot it. It converges to zero slower or faster and it can oscillate around zero. It all depends on your system configuration. You have to decide what is  acceptable. You can read this value for each slave. Watch out, the value is not 2s-complement. Bit 31 0 = local copy is less than received time, 1 = local copy is more than received time.
2. As you can see in the capture, the first FPRW is 2 bytes so it overs both 0x980 and ox981. Btw, not only sync0 is activated but also sync1.
3. How is anyone supposed to be of help here? You do not provide enough data. Write facts that can be checked, not conjectures and intentions. Show the code and the resulting wireshark traces.

nakarlsson, 2023-10-25:
The highest bit is the sign, eg. +/-. So you should disregard that to know that the differnece is 419 ns. Is that acceptable for you, then your good.

I'll close this issue, you have so many going on in parallel.

closed 2023-10-25 by nakarlsson