OpenEtherCATsociety/SOEM #752: how to trigger sync0 signal

Edphoenixmax, 2023-10-17:
I just use the configdc and ecx_sync0 function before slave state change to safeop , but I still cannot change to state OP.
I check the frame from wireshark and I find that the value of ox981 has to be set 00 so that state can be changed correctly, but this means I didn't use sync0 signal to trigger. So is there anything I need to do after calling the ecx_sync0 function to trigger sync0 signal periodicly?
Looking forward to your reply!
Thank you  a lot!

ArthurKetels, 2023-10-17:
Show the code, then I can help. There can be a multitude of reasons why your slave does not want to go to OP.

Edphoenixmax, 2023-10-18:
I am sorry the code is on the desktop that locked , I cannot show you.
Here is the wireshark trace that I can provided for you.
[ethercat-dc07.zip](https://github.com/OpenEtherCATsociety/SOEM/files/12974134/ethercat-dc07.zip)
I will tell you how the code works.
I am trying to download ENI file to do the state change and other configurations just like the codesys does.Here is the ENI file format that I used.
At first the error is 0x002d while these days I donnot know why it has changed 0x001b.
I will appreciated it if you can give me some dvice
Thank you again!
[EtherCAT_Master-configdc.zip](https://github.com/OpenEtherCATsociety/SOEM/files/12974163/EtherCAT_Master-configdc.zip)

Edphoenixmax, 2023-10-20:
Another question , the 0x1c32  0x01 is sync mode , does this value have any relationship with the sync0 or sync1?
I use the slaveinfo eth0 -sdo to check the value , and it is always 1 no matter I call ecx_sync0 function or not.
Anyway ,could you please help me with the question about the wiresharktrace, I would appreciate it that!
I donnot know hoe to express my appreciation that every time I received your reply I was always excited.
You are like a professor , a captain that you can handle every thing. Your experience and your advice are really important to me.
How wish I could have a friend like you around me so that I can expand my horizon.
Thanks again , I am looking forward to your help!

Edphoenixmax, 2023-10-24:
I just check the value of register 0x0982, its DC CycImpulse is 0x0000, is that means it doesn't generate sync0 signal?
Is this corresponded with the slave that I used? I use the weidmuller UR20-FBC-EC coupler and four modules, 16DI-P, 16DO-P, 4AO-UI-16, 4AI-UI-16
Thanks for your help

Edphoenixmax, 2023-10-24:
I just changed the value of DC activation (0x981) 0x02,and it changed state OP correctly while the value is 0x03 , it cannot change into state OP
so What is the difference?
Looking forward to your reply. I would appreciate it very much! Thanks a lot!

nakarlsson, 2023-10-25:
If you don't structure all your open issues into concert issues with meaningful information, closing those not active and so on, we'll close all active issues and give you a new chance to do it the right way. Currently, the Issues written by you are all over the place.

People should take 5 before writing issues, ask them self if they really have understood the problem. Have they tried to find available information on the topic and so forth…

We are only a few people trying to help here, on our spare time. Most issues are not related to SOEM as such, rather on EtherCAT usage and how to implement stuff using EtherCAT.

Edphoenixmax, 2023-10-25:
OK, I get it!
Actrually, you guys are the best person I know who understand Ethercat protcol best!
Thanks again for your help!

closed 2023-10-25 by Edphoenixmax