OpenEtherCATsociety/SOEM #754: slaveinfo cannot return correct size of the last slave's pdo, and can only detect pdo1

hlee-hss, 2023-10-24:
I am trying to use soem to read TxPDO from maxon's ethercat motor driver. However, although I can see the pdo mapping on the motor driver is correct, the input size I get from slaveinfo is always incorrect. In fact, maxon driver has multiple pdo (pdo1, pdo2), but only the first pdo is detected. Is there any way to fix it? Or I need a paid version of soem? I am willing to paid the price if the customer service can reply to me.

nakarlsson, 2023-10-25:
What you see is the default PDO mapping, you would need to setup the desired PDO mapping using for example the PO2SOconfig hook during startup. Several examples of that in the "Issues" section.

This is not SOEM specific, TwinCAT would do the same.

hlee-hss, 2023-10-25:
I guess I misunderstood the meaning of pdo, even I can configure pdo1~pdo4, I can only select one right?

nakarlsson, 2023-10-25:
Often you can select more, consult the UserManual or ESI file. They express if there are any PDOs that are mutually exclusive.

hlee-hss, 2023-10-27:
It turned out the motor driver I use only support single pdo, thank you for all the helps.

closed 2023-10-27 by hlee-hss