OpenEtherCATsociety/SOEM #755: simple_test exits with message "Slave 1 State=0x11 StatusCode=0x0016 : Invalid mailbox configuration"

de6b30, 2023-10-26:
I am trying to connect to a commercial slave device and simple_test can't set the slave to operational state. I suspected that maybe there is a problem with the EEPROM data of the slave in some way and therefore I implemented a static slave configuration in ethercatconfiglist.h. I think this improved the situation already, but I still get the same error message and operational state isn't reached.

This is the complete output of simple_test:

Simple test
Starting simple test
ec_init on enp0s31f6 succeeded.
ec_config_init 1
1 slaves found and configured.
ec_config_map_group IOmap:0x55b53c9b3180 group:0
 >Slave 1, configadr 1001, state 11
     ISIZE:704 704 OSIZE:224
  SM programming
    SM2 Type:3 StartAddr:1800 Flags:00010064
    SM3 Type:4 StartAddr:1c00 Flags:00010020
  OUTPUT MAPPING
    FMMU 0
      SM2
    slave 1 Outputs 0x55b53c9b3180 startbit 0
 =Slave 1, INPUT MAPPING
    FMMU 1
      SM3
    Inputs 0x55b53c9b319c startbit 0
IOmapSize 116
Slaves mapped, state to SAFE_OP.
segments : 1 : 116 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x11 StatusCode=0x0016 : Invalid mailbox configuration

Request init state for all slaves
End simple test, close socket
End program

I also recorded a trace file with wireshark: 
[RTFP_connection_failed.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13174889/RTFP_connection_failed.zip)

Can somebody maybe give me some hint what is going wrong? I am not used to reading EtherCAT network traffic. 

I tried simple_test with another slave before and everything worked fine, so I think I don't have a general problem. This slave example worked out of the the box: https://www.emotas.de/download/ethercat-beispiele

de6b30, 2024-06-21:
Problem has been solved,

closed 2024-06-21 by de6b30

cindralis-electron27, 2025-07-30:
What was the solution?