OpenEtherCATsociety/SOEM #756: Maps PDO to c struct sometimes result in error. (both RxPDO and TxPDO)

hlee-hss, 2023-10-27:
Perhaps this is a very common problem, however, I could not find much information online. 
I have seen many in the issue section use struct to access both TxPDO and RxPDO, I have follow the similar pattern to read/write data to pdo, however, unexpected behavior occurs. 
 
let's say the struct is
```c++
 struct TxPDO1_Data {
        uint16_t status_word;
        uint16_t control_word;
        int32_t position_actual;
        int32_t velocity_actual;
        int16_t torque_actual;
        uint16_t analog_input1;
        uint16_t analog_input2;
        uint16_t digital_input;
        uint8_t modes_of_op;
    };
```
and I copy the TxPDO as:
```c++
TxPDO1_data pdo_data = *(TxPDO1_Data*)ec_slave[slave_id].inputs;
```
While it gets me the correct data. Yet, if I remove the control_word, the signed values after the first two unsigned values (e.g., position_actual), the value will be wrong (seems to be cast as unsigned values). Is there any risk of directly mapping it to a struct? I have verified the input output data of the slave devices are correct with slaveinfo -map. 

Another problem is commanding slaves with RxPDO.
Suppose we also create a struct for the RxPDO, and create a local variable out of it.
```c++
    struct RxPDO1_Data {
        uint16_t control_word;
        uint32_t target_pos;
        uint32_t target_vel;
        uint16_t target_torque;
        uint8_t modes_of_op;
    };

RxPDO1_Data local_rxpdo;
```
Is sending out the local_rxpdo just
```c++
while(true) {
    ec_send_processdata();
    int wkc = ec_receive_processdata(EC_TIMEOUTRET);
    functionUpdateRxPDO();
    std::memcpy(ec_slave[slave_id].outputs, &local_rxpdo, sizeof(local_rxpdo));
}
```
Currently I cannot get the slaves react to any commands. I will like to know if we are using the soem wrong.

nakarlsson, 2023-10-27:
You need to consider alignment and padding.
Often you pack a structure.

hlee-hss, 2023-10-27:
Oh you are correct, padding is the source of the problem. 
All issue resolved after adding 
```c++
#pragma pack(push, 1)
// struct define
#pragma pack(pop)
```

closed 2023-10-27 by hlee-hss