OpenEtherCATsociety/SOEM #757: change to state op

Edphoenixmax, 2023-11-01:
I wish everything goes well.
I set the manualstatechange 1 to set slaves state so that I can control slaves' state before I have already configured them all.
But when I call for ecx_sync0 function but I cannot change into state OP and the error code is 0x002d SyncSignal not received
I have already check the syncmode , the value of 0x982 DC – Pulse length of SyncSignals is always 0 (it seems like the value of this register can be changed when you use the eepromtool to redownload the xmlfile)so that the sync mode is always Cyclic Acknowledge Mode .
So what I can do to set_stateOP successfully?Should I call for the set_state function at the cycletime pace or something else?
In my opinion, manualstatechange=0 is trigerred by SM event, so how can I use the sync0 event to triger the statechange?
Thanks for your help!Best wishes!

ArthurKetels, 2023-11-01:
Hi,as usual you do not provide the data needed to give sensible answers. You obviously are doing something wrong in your code, but you don't want to share that code. 

If you set sync0 to fire at set intervals make sure you set the first trigger far enough in the future. If not, the absolute time has already past and you have to wait for the counter to wrap from zero again (and that can be a very long time for 64bit time). But this is just a very common mistake, there are hundreds of other reasons why it does not work as you intend to.

Have a look at post : https://github.com/OpenEtherCATsociety/SOEM/issues/457#issuecomment-715564569

Edphoenixmax, 2023-11-01:
Thanks for your opinion and help!
I really want to share the code with you so that I can get the advice, but my desktop is locked in the local area network and it cannot transmit any information outside. So I cannot share it with you through the network. I am sorry about that.
You mean that sync0 signal has been triggered before I set stateOP,while it has to wait for another signal trigger .
But why the interval is not the cycletime you set? I don't understand.
I will check the #457 comment carefully.
Thanks again!

closed 2023-11-06 by Edphoenixmax