OpenEtherCATsociety/SOEM #758: Slave wont go to SAFE OP instead goes to PRE OP

IceTea2, 2023-11-01:


closed 2023-11-02 by IceTea2

prasaanthLvk, 2024-01-19:
Hi.., @IceTea2 Can you please let me know how you solved this ?I am also getting the same error the Device is not going to Safe_op and is showing error.If you could help me out it would really be a great help to me.

IceTea2, 2024-01-19:
> Hi.., @IceTea2 Can you please let me know how you solved this ?I am also getting the same error the Device is not going to Safe_op and is showing error.If you could help me out it would really be a great help to me.

Which device do you have exactly? in my case, i needed to assign the pdo mapping via sdo_write for both receiving and transmitting pdo.
And that right before going to safe op

prasaanthLvk, 2024-01-20:
Device :: ClipX BM40IE from HBM 

Thank you so much,for your info

prasaanthLvk, 2024-01-21:
Hi @IceTea2 is it possible for you to share your pdo mapping I have tried mapping my pdo but it is failing with both my and soem example code stating that it's a read only not   writable