OpenEtherCATsociety/SOEM #759: How to verify that the PLL of the master has locked

Edphoenixmax, 2023-11-08:
I wish all you guys are good!
I have already found out why the error occurred 
Is there any way to solve this error through SOEM functions?
 
[image]

this is the description:
ESM Timeout SafeopOp is used
Slave application verifies if SM-Event occures before Sync0-Event for a defined number of cycles per time window for the complete SafeopoP-Timeout-Time
Ok: Confirm State Change to OP
nOk: AL Status = 0x14 | AL Status Code = 0x2D (No Sync)

Thanks for your reply!

ArthurKetels, 2023-11-08:
Look at the source of red_test. This is an example of master synchronization. Actually it is outside the scope of SOEM. You as an application programmer are responsible for timing. Basic requirements are a low latency OS, good timing accuracy and a simple feed back loop between the slave reference clock and you main PDO loop. 

Do read the manual of your slave to figure out the requirements with respect to the SM offset vs the SYNC0/1 event.  A good start is half way in between the SYNC0 cycle.  I have written multiple post here on what steps to take and how to check each step.

Edphoenixmax, 2023-11-09:
OK, I get it
Thanks a lot for your advice!

closed 2023-11-09 by Edphoenixmax