OpenEtherCATsociety/SOEM #760: get EMCY values

open04, 2023-11-14:
Hi, Is there a way to get the Emergency Messages(EMCY) values?
Thanks

ArthurKetels, 2023-11-14:
They are automatically handled by the mailbox receive function. Any time ecx_mbxreceive() is called it checks if an emergency message is received, if so it puts it in the error list. This can be checked by ecx_iserror(), and retrieved by ecx_poperror(). 

It is up to the master application to setup a mechanism to read slave mailboxes at appropriate times. The most efficient way is to program an FMMU on the slave to mirror the mailbox SM state in the PDO mapping.

As a separate remark, it is not recommended for slave manufacturers to use the emergency mailbox protocol. It is inefficient and can be difficult to implement with regards to real-time requirements. This is legacy CANopen stuff, slave errors should be handled in the PDO mapping.

CAcarADI, 2023-11-15:
Thanks Arthur for you reply, Ill try it anyway. Ill just discard it if it does not fit on my application.

open04, 2023-11-15:
Thanks Arthur for the reply, Ill give it a try still. I will just discard it if it does not fit in my application

closed 2023-11-15 by open04