OpenEtherCATsociety/SOEM #763: print ECAT frames

open04, 2023-11-22:
Is it possible to print the ECAT frames?  will use for debugging purposes. I want to print out the frames when I write on SDO or PDO.

ArthurKetels, 2023-11-22:
Yes, easiest is to use WireShark. It captures all Ethernet traffic, and has special filters for EtherCAT build in.

open04, 2023-11-22:
Hi Arthur, sorry for confusion but what I meant was to print the ECAT frame in the code itself, in Cpp. I want to see the frames Im sending in the terminal, not to use another application.

ArthurKetels, 2023-11-23:
You can do it yourself too. If your print a hexdump of the outgoing and incoming packets. Or you make a complete parser like WireShark does. Why can't you simply use WireShark? Everyone else does it this way with great satisfaction.

The location you want to plug into are ecx_outframe() and ecx_waitinframe().

open04, 2023-11-24:
Thanks Arthur!, Ill try the ecx_outframe() and ecx_waitinframe().

Im using Wireshark and I can also read frames from there. What I want is to print selected frames on the terminal which makes it easy for me and also my colleague to debug if we encounter problems.

open04, 2023-11-28:
I didnt expect it to be complicated. I will not include it on my package, Ill close this issue now.
Thanks again Arthur

closed 2023-11-28 by open04