OpenEtherCATsociety/SOEM #764: Cannot transit to OP state with Veichi servo drive

NevolinAlexander, 2023-12-02:
Hi everyone! I'm newbie in ethercat. Earlier I used SOEM to work with delta servo drives and everything went fine. But now I am trying to work with servo drive Veichi SD700. It's base on ESC Beckhoff ET1100. And even in simple_test it cannot reach operational state and returns no error:

Slaves mapped, state to SAFE_OP.
segments : 1 : 40 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

As I read issues here - some ESC require specific work with. But most of them are about TI ESCs, and I have (maybe) classical beckhoff slave. Can you advise me something to check or try? Thanks in advance!

NevolinAlexander, 2023-12-04:
Thank you very much for your reply. 

How should I write mentioned objects correctly, before assiging PDO procedure or after it?

What I'm doing know:
1. Entering PRE_OP
2. Assigning PDO
3. Writing 0x6060, 0x60c2 (values 0x08, 0x01, 0xFD - all 8 bit)
4. Calling ec_dcsync0
5. Calling ec_config_map
6. Calling ec_configdc
7. Entering SAFE_OP
8. Do 25 async cycles delay
9. Trying to enter OP

Using this steps drive shows error on it's indicator - "SYNC0 sync frame is lost".

And what text "<AssignActivate>#x300</AssignActivate>" means? Should I write something additional to activate sync0 at slave?

ArthurKetels, 2023-12-05:
Remarks:
3. Check if written values can be read back and are correct.
4. ec_dcsync0 can not be written before ec_configdc, so move it after 6.

Position 8 should be done very carefully. The best thing to do is starting synchronous PDO cycles after ec_configdc.  Have a control loop in place that locks the PDO cycle with the reference slave clock. Only when the cycle is locked and stable with regard to the reference clock transition to safe-OP. In safe-OP wait for some time, some slaves need more than a second to detect a stable relation between PDO cycle and sync0.

The "#x300" is the value written to register 0x0980 (Cyclic Unit Control) and 0x0981 (Sync out unit). It sets the bit for sync unit activation and sync0 generation.

NevolinAlexander, 2023-12-05:
Thank you for reply! 

I've checked how values are read and found that 0x60c2 are 16-bit values, because after writing as 8-bit they didn't read correctly. I fixed this issue and now all of them have needed value after reading.

So how I have changed my code:

1. Assigning PDO, writing 0x606, 0x60c2
2. Call ec_config_map
3. Call ec_configdc
4. Call ec_dcsync0
5. Make 10 synchronous cycles
6. Transit to SAFE_OP
7. Making 1000 more synchronous cycles (with PI-regulator that is in code)
8. Request OP.

But slave does not enter this state and shows no any error. Display on drive shows digit "4" that means that it's in SAFE_OP. 

Also when program exits (and does not send any packets anymore) the drive continues to show "4" and does not show any error code like earlier (I wrote it in previous message).

NevolinAlexander, 2023-12-09:
Thank you Arthur again for showing vector of search. Damned chinese drives.
Actually it requires sync0 before safe_op. My problem was that I didn't know that slaves mapping function automatically does transition to safe_op. This lead to safe_op state before sync0 starting, and it was the problem. Now drive successully goes to the OP state.

Another thing that confused me was that after shutting down of SOEM application drive showed on it's display that it is still in safe_op. Due to Ethercat spec can slave stay in safe_op state without any data exchange from master? Or should it fall in error?

Also if drive works in OP state and SOEM app exits, drive fall in appropriate error, but after it I can't start it again - SDO write in PRE_OP fails. Only power reset helps. Do I have to make any special procedure to recover it? There is nothing in manual about it. Other drives do not have such issue.

And one more interesting thing is that drive is not moved to ENABLE_OPERATION state when trying to do servo_on. It stucks on  SWITCH_ON state. But it's fully functional, accepts all modes (cyclic position/velocity/torque) and everything works fine. Funny thing for me.

ArthurKetels, 2023-12-09:
Welcome to the wonderful world of drive developers. If have seen so many EtherCAT products that really do not have the best user experience in mind. Many of them fail the EtherCAT standard, others interpret them in "curious" ways. I am both a master as well as slave developer, so I see both worlds.  I always strive to develop a product that is as forgiving as possible to common user errors, and allow multiple views of what is the right thing to do. I guess not all of us do that in the same way. My recommendation is to try to talk to vendors about firmware quality, and when that does not help, stay away, choose another vendor.

Handling failures of any kind in a graceful and safe way is also a subject of much discussion. For a slave being stuck in a state after a failure is not acceptable to me. Can you trust it to perform correctly in other situations?

NevolinAlexander, 2023-12-09:
The most interesting thing is that this Veichi drive uses Beckhoff IC as a ESC (ET1100). I thought that if they use it, this drive must follow ethercat standard (because Beckhoff is it's author). Or this states logic is implemented at higher level?
All other drives that we used before, like Delta, implement ESC in other ways, for example using Altera Cyclone's.

Talking about most errors forgiving drive, this is about Delta drives. Working on this issues, I discovered few errors in our master implementation (violating standard), but their drives work well.

I fully 100% agree with your big comment about synchronisation (mentioned many times here) - and I will never buy Veichi drives more. Just as principle. 

And of course drive should not be stuck in error, you are right. Maybe there is some procedure to recover it, but there is no any info in their manual.

NevolinAlexander, 2023-12-09:
So why do you think Beckhoff changed Twincat behaviour and began to start sync0 in pre_op state? Maybe they were just tired to get questions - why Twincat does not work with some weird slaves? In fact it lead to vicious circle - ESC vendors test them only in Twincat (as most popular platform), it works OK and they think it's enough.

ArthurKetels, 2023-12-11:
Use of a specific ESC chip does not tell anything about the quality of the firmware stack. Anything more than registers, SM and FMMU is handled by software. Even using a validated EtherCAT firmware stack (like SOES for example) does not tell you much about the application layer on top of that. Do you know how many cycles delay (if any) are there between the EtherCAT packet reception and the actual change in the actuator? You would be surprised with some vendors.

Twincat has had some mayor shifts in operation over time. Mostly to solve specific customer issues. But then this behaviour kind of sets the reference. Indeed most slave vendors test their product with twincat, and that is it. There is also a mandatory slave conformance test from the ETG, but that does not cover the application behaviour above the EtherCAT stack. And it is exactly there where most issues are.

Demanding stable synchronized communication with Sync0 in pre-OP is ridiculous. The most sensible spot for a slave to detect that is in the transition from safe-OP to OP. And only for very specific input critical slaves (like encoders), the transition from pre-OP to safe-OP.