OpenEtherCATsociety/SOEM #765: some slaves that donnot support sync signal trigger

Edphoenixmax, 2023-12-04:
I wish you all good!
Recently I use the ecx_dcsync0 function to trigger the servomoto successfully while the same code cannot trigger weidemuler slave like UR20_4FC_EC , Is there any possibility that the digital IOmoudles donnot support the sync signal trigger?

ArthurKetels, 2023-12-05:
Some more information would be welcome. Can you supply the ESI file of the UR20? There you can read the supported modes. Many slaves only support SM synchronization.

Edphoenixmax, 2023-12-05:
[WI_ESI_HW2.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13557439/WI_ESI_HW2.zip)
Here is  the ESI of UR20 , where can I know the slaves support only SM synchronization?
Thanks for your help!

ArthurKetels, 2023-12-05:
I could not find the UR20-4FC-EC but you have to look for something like:
```
				<Dc>
					<OpMode>
						<Name>SM_SYNCHRON</Name>
						<Desc>FreeRun/SM Synchronous</Desc>
						<AssignActivate>#x0000</AssignActivate>
					</OpMode>
					<OpMode>
						<Name>DC_SYNCHRON</Name>
						<Desc>DC Synchronous</Desc>
						<AssignActivate>#x0300</AssignActivate>
						<CycleTimeSync0 Factor="1">0</CycleTimeSync0>
					</OpMode>
				</Dc>
```
This snippet was from the UR20-FBC. And it shows support for both SM and Sync0.

Edphoenixmax, 2023-12-05:
Sorry , I forget its name. Its real name is UR20-FBC-EC, but it did support the sync0 and I donnot know why it cannot change into op station. It is so weird.

closed 2023-12-14 by Edphoenixmax