OpenEtherCATsociety/SOEM #770: Beckhoff power supply DC error

ahmetihsankose, 2023-12-11:
Hello, I'm trying to get Beckhoff slaves up and running using SOEM EtherCAT master. I've attached the sample code, Wireshark data, and ENI file. I've noticed an issue where the power supply slave in the first position returns wkc == -5 in the ecx_mbxreceive function. This might be causing problems in setting up the DC. According to the current state of the provided code, the output is like this, and I can't switch to the SafeOP state:

Slave:1 Name:AX8640-0000-0103 Output size:  0bits Input size: 16bits State:18 delay:0.1
         Out:(nil),   0 In:0x55a44b4c436c,   2
Slave:2 Name:AX8206-0000-0103 Output size:176bits Input size:176bits State: 4 delay:175.1
         Out:0x55a44b4c4340,  22 In:0x55a44b4c436e,  22
Slave:3 Name:AX8206-0000-0103 Output size:176bits Input size:176bits State: 4 delay:390.1
         Out:0x55a44b4c4356,  22 In:0x55a44b4c4384,  22
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  30 : Invalid DC SYNC configuration
End program

Also, I should mention that, when I try to bring up the drivers for the first time after giving power, I get the following output. However, afterward, the error I mentioned above becomes constant.

Slave:1 Name:AX8640-0000-0103 Output size:  0bits Input size: 16bits State:18 delay:0.1
         Out:(nil),   0 In:0x559ac2355340,   2
Slave:2 Name:AX8206-0000-0103 Output size:  0bits Input size:  0bits State: 1 delay:169.1
         Out:(nil),   0 In:(nil),   0
Slave:3 Name:AX8206-0000-0103 Output size:  0bits Input size:  0bits State: 1 delay:389.1
         Out:(nil),   0 In:(nil),   0
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  30 : Invalid DC SYNC configuration
Slave 2 State= 1 StatusCode=   0 : No error
Slave 3 State= 1 StatusCode=   0 : No error
End program
[test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13637135/test.zip)

Thanks for everything.

ArthurKetels, 2023-12-11:
There are multiple problems:
1. Slave 1 does not respond to the SDO upload request 0f 0x1c00:00
2. Your application start cyclic SDO at safe-OP to OP transition. This needs to be much earlier at pre-OP just before Sync0 activation.

Can you show the output of slaveinfo -map? Also the ESI file of the AX8640 would help (and please cut it to only this slave, I have no intention to walk through hundreds of slaves).

ahmetihsankose, 2023-12-12:
I will configure my code based on your first suggestion. It looks like it's entering SafeOP Mode properly right now, but I still can't transition to Operational Mode but I'm still facing a persistent wkc -5 error from the ecx_mbxreceive function for the first slave.
I have also attached my fixed code, and the attached ESI file contains only the part with the AX8640 slave.

According to the new code, the output is as follows.

Slave:1 Name:AX8640-0000-0103 Output size:  0bits Input size: 16bits State: 4 delay:0.1
         Out:(nil),   0 In:0x564ba35d836c,   2
Slave:2 Name:AX8206-0000-0103 Output size:176bits Input size:176bits State: 4 delay:170.1
         Out:0x564ba35d8340,  22 In:0x564ba35d836e,  22
Slave:3 Name:AX8206-0000-0103 Output size:176bits Input size:176bits State: 4 delay:385.1
         Out:0x564ba35d8356,  22 In:0x564ba35d8384,  22
Calculated workcounter 7
Request operational state for all slaves
Not all slaves reached operational state.
Slave 1 State=0x00 StatusCode=0x0000 : No error
Slave 2 State=0x00 StatusCode=0x0000 : No error
Slave 3 State=0x00 StatusCode=0x0000 : No error
Request safe operational state for all slaves

slaveinfo -map output is as follows

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enx207bd24f48a0 succeeded.
3 slaves found and configured.
Calculated workcounter 7
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  30 : Invalid DC SYNC configuration
Slave 2 State=12 StatusCode=  30 : Invalid DC SYNC configuration
Slave 3 State=12 StatusCode=  30 : Invalid DC SYNC configuration

Slave:1
 Name:AX8640-0000-0103
 Output size: 0bits
 Input size: 16bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 21c06012 Rev: 00670000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1100 L: 256 F:00010022 Type:2
 SM2 A:1200 L:   0 F:00000004 Type:3
 SM3 A:1900 L:   2 F:00010020 Type:4
 FMMU0 Ls:0000002c Ll:   2 Lsb:0 Leb:7 Ps:1900 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 27 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
ecx_mbxreceive: wkc = -5
ecx_mbxreceive: wkc = -5

Slave:2
 Name:AX8206-0000-0103
 Output size: 176bits
 Input size: 176bits
 State: 18
 Delay: 170[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 200e6012 Rev: 00670000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
[... 69 lines cut ...]
  [0x0022.0] 0x6840:0x00 0x10 UNSIGNED16   Ch B Controlword
  [0x0024.0] 0x687A:0x00 0x20 INTEGER32    Ch B Target position
  [0x0028.0] 0x68FF:0x00 0x20 INTEGER32    Ch B Target velocity
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0044.0] 0x0000:0x00 0x01
  [0x0044.1] 0x603E:0x02 0x01 BOOLEAN      Position actual value invalid
  [0x0044.2] 0x603E:0x03 0x01 BOOLEAN      1st Additional position actual value inv
  [0x0044.3] 0x0000:0x00 0x05
  [0x0045.0] 0x6041:0x00 0x10 UNSIGNED16   Ch A Statusword
  [0x0047.0] 0x6064:0x00 0x20 INTEGER32    Ch A Position actual value
  [0x004B.0] 0x60F4:0x00 0x20 INTEGER32    Ch A Following error actual value
  [0x004F.0] 0x0000:0x00 0x01
  [0x004F.1] 0x683E:0x02 0x01 BOOLEAN      Position actual value invalid
  [0x004F.2] 0x683E:0x03 0x01 BOOLEAN      1st Additional position actual value inv
  [0x004F.3] 0x0000:0x00 0x05
  [0x0050.0] 0x6841:0x00 0x10 UNSIGNED16   Ch B Statusword
  [0x0052.0] 0x6864:0x00 0x20 INTEGER32    Ch B Position actual value
  [0x0056.0] 0x68F4:0x00 0x20 INTEGER32    Ch B Following error actual value
End slaveinfo, close socket
End program

Thank you for your interest

[test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13646289/test.zip)

ahmetihsankose, 2023-12-29:
Hello again, first of all, I realize I didn't send the attached file correctly earlier, so I'm resending it. I'm still getting errors in the initial configuration for the power supply. However, apart from this, I'm not receiving any other errors. 
The issue of the txpdo values being zero, which I mentioned earlier, seems to be due to my incorrect configuration, but I've resolved that now. In addition to this, for the motor to be active among these slaves, it appears that slave-to-slave communication is necessary. I believe I'm not doing this synchronization correctly. Could you assist me with this?
I had arranged the code according to the writing and reading in TwinCAT when I logged the input and output data in the linkage between this power supply and the other two slaves in TwinCAT.
Thank you for everything, I realize I've gone a bit off-topic. Besides, if that's the case, we can close this topic.

```
void tick()
{
    if (runControlFlag > 0)
    {

        WorkingCounter = ec_receive_processdata_group(0, 2000);
        ecx_send_overlap_processdata_group(0);

        ExpectedWorkingCounter = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
        if (WorkingCounter < ExpectedWorkingCounter)
        {
            std::cout << "WKC ERROR, expected " << ExpectedWorkingCounter << ", got " << WorkingCounter << std::endl;
        }

        if (currentState == EC_STATE_OPERATIONAL)
        {
            previousReadValue = currentReadValue;
            currentReadValue = get_input_uint16(1, 0, 2); // first 2 byte for 1st slave
            set_output_uint16(2, 0, 2, previousReadValue); // first 2 byte for 2nd slave
            set_output_uint16(3, 0, 2, previousReadValue); // first 2byte for 3rd slave
        }
    }
}
```
[AX8640-0000-10000.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13792488/AX8640-0000-10000.zip)

ahmetihsankose, 2023-12-31:
After examining the Wireshark data in more detail, I noticed that the writing of the output data, which is supposed to be based on the input data, is occurring before the input data is read. This suggests to me that this data is written before it's actually read by the master. From some of my research, I think this can be achieved using FPRD/APRD and FPRW/APWR, which means my previous implementation seems to be completely incorrect. Hopefully, I will have tested this within two days and will write about the results.

ArthurKetels, 2024-01-01:
Unfortunately the TwinCAT trace has no configuration part. It starts when the slaves are already in OP. So it is no use to figure out how TwinCAT set-up the slaves. If you can add that part.

ahmetihsankose, 2024-01-02:
> Unfortunately the TwinCAT trace has no configuration part. It starts when the slaves are already in OP. So it is no use to figure out how TwinCAT set-up the slaves. If you can add that part.

Arthur, thank you very much for your interest. The startup's Wireshark data is attached.

I have an additional issue with these slaves working modularly. When looking at TwinCAT, it seems that the first two bytes of the input of the first slave, which is a power supply, are linked with the first two bytes of the second and third servo slaves. However, when I examine this through Wireshark data, I strangely observe that the output data is captured before the input data. I find it hard to understand how it can write to the output without reading the input first.

[startup.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13811167/startup.zip)

ArthurKetels, 2024-01-03:
The start-up of TwinCAT is quite peculiar. At least the SOEM configuration of SM and FMMU is the same for all slaves. TwinCAT starts synchronizing clocks as soon as Init state. It also regularly reads the CRC-error registers of the slaves during state transition to check for communication instabilities. Each slave is brought up through all states independently. So one slave could be in OP while others are still in pre-OP.

All 3 slaves do support CA (Complete Access) mode for SDO. So I think the -5 (timeout) error SOEM gives after reading SDO 0x1c00:00 CA of slave 1 is still not understood. It could be that the timeout has to be increased.

You observation that the command word of the PSU slave needs to be copied in the PDO's of the drive slaves is correct. This is simply to inform the drives of the intentions, and state of the PSU. If there is for example an overcurrent event, the PSU commands all drives to back down in power consumption. The drives can check the validity of the PSU command by checking a 5 bit incrementing counter that is also in the command word.

The TwinCAT trace is indeed a bit confusing with respect to the ordering of the packets. But rest assured the data is first read before it is written. It is just that TwinCAT does not place the packets to Wireshark in chronological order. You can check this by looking at the index field of a datagram. This is set by the master and is incremented every next cycle. The return packet from the slaves have the same index. And what you see in the Wireshark trace is that the received packet from the slaves is put before the transmitted packet.

This effect could be caused by having two separate buffers to copy packets to the windows NDIS stack. One buffer for Tx packets and one buffer for Rx packets. And each buffer runs in its own task. Remember that TwinCAT runs separate from Windows in its own real-time OS.

ahmetihsankose, 2024-01-04:
> The start-up of TwinCAT is quite peculiar. At least the SOEM configuration of SM and FMMU is the same for all slaves. TwinCAT starts synchronizing clocks as soon as Init state. It also regularly [...]

Thank you for the detailed and informative response. Currently, I am facing an error code 12928. When I look into the documentation, it says 'DC link is not ready.' I am suspecting this might be due to the maximum DC voltage setting in the power supply's startup configurations.

What I am curious about is that when I check with slaveinfo, only FMMU0 is visible for the PSU. However, when I look into the ENI file, it appears that settings for FMMU1 are also being configured for the PSU. Could this be a reason for the issue? Also, would it be worthwhile to parse the initcmnds in the ENI and try to bring up the slave using SOEM, or is this not a recommended approach?

When I set longer waiting times for SDOs, -5 wkc is still remain.
[ENI_BECKHOFF_4ms.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13829946/ENI_BECKHOFF_4ms.zip)

lshoek, 2024-04-24:
> Thank you for the detailed and informative response. Currently, I am facing an error code 12928. When I look into the documentation, it says 'DC link is not ready.' I am suspecting this might be due to the maximum DC voltage setting in the power supply's startup configurations.

Hello, did you manage to solve this? I am currently running into the same problem with these drives. More specifically, I get the following two:
- [0x5187, Power supply communication is not established](https://infosys.beckhoff.com/english.php?content=../content/1033/ax8xxx_diags/11593008011.html&id=)
- [0x3280, DC link is not ready](https://infosys.beckhoff.com/index.php?content=../content/1031/ax8xxx_diags/11592999819.html&id=)

All devices are in EtherCAT OP state. These errors come up when trying to advance the DS402 state machine from `ReadyToSwitchOn` to `SwitchOn`. When I try this the next frame's status word is a `Fault`. TwinCAT 3 on the other hand seems to be able to do this effortlessly, even though the startup configuration is identical to my custom master. The first six bits of the power supply control word for axis are always high:
```
[0x0018.0] 0x6000:0x01 0x01 BOOLEAN Non-generative brake order_n
[0x0018.1] 0x6000:0x02 0x01 BOOLEAN Ready to operate
[0x0018.2] 0x6000:0x03 0x01 BOOLEAN TorqueOFF order_n
[0x0018.3] 0x6000:0x04 0x01 BOOLEAN Generative brake order_n
[0x0018.4] 0x6000:0x05 0x01 BOOLEAN Limit dc link current_order_n
[0x0018.5] 0x6000:0x06 0x01 BOOLEAN NC handling_order_n
```
The counter seems inconsistent when I log the bits however? As in, it seems to increment more than one each frame. Not sure if this is even a problem at all. I just copy the `UINT16` to the appropriate location of the axis output each frame. Thanks and please lmk!

ahmetihsankose, 2024-04-25:
Unfortunately, I had to leave it unresolved as I couldn't find a solution, and I'm encountering the same errors as you. When trying to write SDO settings for my power supply slave during initial configuration, I was getting a WKC of -5. I think we're seeing these (0x3280) error codes because the settings might not be entered correctly.

LukasW-Dev, 2024-07-11:
> Hello, did you manage to solve this? I am currently running into the same problem with these drives. More specifically, I get the following two:
>
> * [0x5187, Power supply communication is not [...]

Hello, I have the exact same problem. Did you manage to find a solution?

Guistac, 2025-12-10:
Hi @lshoek,
this was indeed what was missing from my configuration ! Thank you so much for your insight !

Upon resolving this issue, i was faced with some others:

> - 0x7398, Physical position sensor required
> - 0x8A81, Illegal Modes Of Operation Change

I'm using a 2 Channel AX8206 drive for OCT/Hiperface Beckhoff motors only.
As it turns out, the slots for the feedback module of each drive channel are not populated by default. The servo drive explicitly requires that configuration to happen during the IP transition. (Init to Preop)

This is the relevant excerpt from my ESI file:

`<Slots DownloadModuleIdentList="1" DownloadModuleListTransition="IP" MaxSlotCount="64" MaxSlotGroupCount="2" SlotPdoIncrement="0" SlotGroupPdoIncrement="0" SlotIndexIncrement="0" SlotGroupIndexIncrement="0">`

The ETG.2000 Document about ESI files states the following:

> If transition is ‘IP’ the command is send directly after the slave reached PreOp state .

As if that wasn't annoying and confusing enough, the drive rejects setting only the subindex of the required slot, you have to upload the entire object using an SDO write with CA (complete access). So I ended up adding the following routine to my startup process:

- Read the entire object 0xF050 with CA
- You now have all modules that are configured by default (except the encoder related ones)
- Modify the result by adding the missing modules (you can find the right 32 bit values in twincat, under device slots)
- Overwrite the entire object 0xF030 with the updated list

```
struct ModuleList{
	uint8_t size;
	uint8_t padding;
	uint32_t modules[96]; //adjust this to your module list size
}__attribute__((packed));

ModuleList list;
int size = (int)sizeof(List);
int wkc = ec_SDOread(slaveIndex, 0xF050, 0x0, true, &size, &list, EC_TIMEOUTSAFE);

//set configured module in slot 55 and 87 to OCT rotary (hiperface DSL) == 0x6A59D9
list.detectedModules[55] = 0x6A59D9;
list.detectedModules[87] = 0x6A59D9;
wkc = ec_SDOwrite(slaveIndex, 0xF030, 0x0, true, sizeof(List), &list, EC_TIMEOUTSAFE);
```

After doing all of this, i could finally enable the axis power stages and move my actuators.

One issue remains with my drives: after getting my drives to OP I cannot do SDO downloads, even after stopping the cyclic exchange. The drive just floods back a bunch of CoE Emergency messages and the transfers fail. It does this even after setting object 0x10F3.5 to not send emergency messages. I've commissioned about 30 different EtherCAT SubDevices with SOEM and it is the only one that behaves like this, very annoying...

ArthurKetels, 2026-05-10:
@Guistac , Iwould be interested in your solution. Perhaps I can help others.
The OEMC project is released (at least internal at TU/e). I have a look if it is accessible via the web.