OpenEtherCATsociety/SOEM #771: I tried to port SOEM to STM32F407, but the RAM overflowed

fyyxxm, 2023-12-12:
STM32F407 only has 128K RAM, it seems that the RAM used by SOEM is much larger than 128K. How can I modify the configuration of SOEM to reduce RAM usage?

nakarlsson, 2023-12-12:
Decrease number of frames, EC_MAXBUF.

We’ve run SOEM in 48k.

fyyxxm, 2023-12-13:
> Decrease number of frames, EC_MAXBUF

Thank you for your reply. I will make the necessary modifications and try again

nakarlsson, 2023-12-13:
You can also decrease the number of pre-allocated slaves,

/** max. number of slaves in array */
#define EC_MAXSLAVE       200

fyyxxm, 2023-12-14:
> You can also decrease the number of pre-allocated slaves,
> 
> /** max. number of slaves in array */ #define EC_MAXSLAVE 200

Thanks, after trying both of these modifications, they are effective. Thank you for your help.

closed 2023-12-14 by fyyxxm