OpenEtherCATsociety/SOEM #773: ESI xml file to Bin File

Lambo0604, 2023-12-18:
Hi~  Is there a tool or conversion rule for converting ESI XML to bin？ 
Now TwinCAT is not stable sometimes;

ArthurKetels, 2023-12-19:
I do not know what you mean with "bin"?

Lambo0604, 2023-12-20:
we can use SSC  to generate  ESI xml file,  and use eepromtool -w bin.file to write it to eeprom;
How to convert  ESI xml to bin.file？

ArthurKetels, 2023-12-20:
Thanks for the clarification. Up to now there are no tools in the SOEM or SOES repository to convert ESI files to eeprom bin files.

The document describing the slave eeprom structure is ETG2010, that you can download from the ETG website.

Lambo0604, 2023-12-20:
Thanks a lot!

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/20/2023 15:15 |
| To | OpenEtherCATsociety/SOEM ***@***.***> |
| Cc | Lambo0604 ***@***.***>,
Author ***@***.***> |
| Subject | Re: [OpenEtherCATsociety/SOEM] ESI xml file to Bin File (Issue #773) |

Thanks for the clarification. Up to now there are no tools in the SOEM or SOES repository to convert ESI files to eeprom bin files.

The document describing the slave eeprom structure is ETG2010, that you can download from the ETG website.

closed 2023-12-25 by Lambo0604