OpenEtherCATsociety/SOEM #774: Whether Soem supports ETG5001

Guiling-zhe, 2023-12-18:
Whether Soem supports ETG5001？

ArthurKetels, 2023-12-19:
ETG5001, modular device profile, is a set of rules for slaves how to manage multiple sub devices in one unit. For the master this has no consequences. SOEM does not care, and should in principle work with every MDP slave.

In other words, your question is not in scope for SOEM. Although it could be relevant for the application you make -with- SOEM.

Thinbungee, 2024-09-16:
Hello Guiling,  SOEM is an excellent example EtherCAT Maindevice (thanks Arthur !) but having modified it to work with a Modular Device (In my case the Wellauto SM877) I think it is unlikely that your ETG5001 device will work with SOEM directly.  Additional steps are required in the mapping of TxPDO and RxPDO and the modules in the modular Sub Device require further steps to be configured.