OpenEtherCATsociety/SOEM #775: Questions about using function ecx_readPDOmapCA（）

fyyxxm, 2023-12-27:
If this function is Thread_n=0,
Then when running to: ecx_readPDOassignCA（）,
Although the inputs are: ECT_SDO_RXPDOASSIGN (0x1c12), ECT_SDO_TXPDOASSIGN (0x1c13).
However, since Thread_n=0, the data will only be stored in context->PDOassign[0], which means that there will only be one context->PDOassign, and the previous ones will be overwritten.
(Only the 0x1A00 mappings will be saved, 0x1600 will be overwritten).
[image]

Is this reasonable?

ArthurKetels, 2023-12-31:
Thanks for a deep look into the code. It is useful to be able to understand the inner workings when having to build an application.

You are indeed right that there is only one store for both the TX and RXPDOASSIGN. But we only need the the total length for configuring the SM, and that is saved separately. How the individual elements are defined is not needed.

However, if your application needs this information you are free to call ecx_readPDOassignCA() again with your own buffer. You need to remember that using any of the functions in ethercatconfig.c is optional. SOEM is just a library, and if you have better means of configuring the slaves than you are free to do that.

fyyxxm, 2024-01-02:
Thank you for your help.
I have added some code according to the routine and implemented the functions I need

closed 2024-01-02 by fyyxxm