OpenEtherCATsociety/SOEM #777: Is ec_slave.output PDO?

Mark9999999, 2023-12-29:
If I want to send RxPDO, my programs are following:

      *(ec_slave[0].outputs) = 0xf;
      *(ec_slave[0].outputs+1) = 0xf;
      *(ec_slave[0].outputs+2) = 0xf;
      *(ec_slave[0].outputs+3) = 0xf;
      *(ec_slave[0].outputs+4) = 0xf;
      *(ec_slave[0].outputs+5) = 0xf;
      *(ec_slave[0].outputs+6) = 0xf;
      ec_send_processdata();
      ec_receive_processdata(EC_TIMEOUTRET);

The outputs are RxPDO, map to "0x6040   0x607A   0x60B1   0x60B2   0x60FF   0x6071   0x6060"
      (The mapping info is from "slaveinfo.c")
      SM2 outputs
          addr b      index: sub bitl data_type    name
      [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
      [0x0002.0] 0x607A:0x00 0x20 INTEGER32    Target position
      [0x0006.0] 0x60B1:0x00 0x20 INTEGER32    Velocity offset
      [0x000A.0] 0x60B2:0x00 0x10 INTEGER16    Torque offset
      [0x000C.0] 0x60FF:0x00 0x20 INTEGER32    Target velocity
      [0x0010.0] 0x6071:0x00 0x10 INTEGER16    Target torque
      [0x0012.0] 0x6060:0x00 0x08 INTEGER8     Modes of operation
      [0x0013.0] 0x0000:0x00 0x08
      [0x0014.0] 0x0000:0x00 0x00
      [0x0014.0] 0x0000:0x00 0x00

But the progams can not work, it can't change the value in my debugging software.
wireshark capture the frame is following

[image]

nakarlsson, 2023-12-30:
Try slave number =1, as in samples

Mark9999999, 2023-12-31:
> Try slave number =1, as in samples

This still doesn't work, my programs are following, Can you help me check it, thank you very much.

`#include <stdio.h>
#include <sys/time.h>
#include <unistd.h>

#include "ethercattype.h"
#include "nicdrv.h"
#include "ethercatbase.h"
#include "ethercatmain.h"
#include "ethercatdc.h"
#include "ethercatcoe.h"
#include "ethercatfoe.h"
#include "ethercatconfig.h"
#include "ethercatprint.h"

char IOmap[4096];
// 使从站进入op状态

void slavetop(int i)
{
   ec_slave[i].state = EC_STATE_OPERATIONAL;
   ec_send_processdata();
   ec_receive_processdata(EC_TIMEOUTRET);

   printf("EC_STATE_OPERATIONAL=%d\n", EC_STATE_OPERATIONAL);
   printf("ec_slave_state=%d\n", ec_slave[i].state);
   ec_writestate(0);
   ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);

}

void simpletest(char *ifname)
{
   int i, k = 1000,j=0;
   int wkc;
   if(ec_init(ifname))
   {  
      printf("start ethernet at %s\n",ifname);
      if ( ec_config_init(FALSE) > 0 )
      {  
         
         printf("found %d slave on the bus\n",ec_slavecount);
         ec_config_map(&IOmap);

         ec_configdc();

         ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);

         for(i=0;i<ec_slavecount;i++)
         {  
            printf("slave%d to op\n", i);
            usleep(1000000);
            slavetop(i);
            printf("i=%d\n", i);
         }  
         
         usleep(300000);

         if(ec_slave[j].state == EC_STATE_OPERATIONAL)
         {  
            while(k--)//延迟
            {  
               
               *(ec_slave[0].outputs) = 0xf;
               *(ec_slave[0].outputs+1) = 0xf;
               *(ec_slave[0].outputs+2) = 0xf;
               *(ec_slave[0].outputs+3) = 0xf;
               *(ec_slave[0].outputs+4) = 0xf;
               *(ec_slave[0].outputs+5) = 0xf;
               *(ec_slave[0].outputs+6) = 0xf;
               ec_send_processdata();
               ec_receive_processdata(EC_TIMEOUTRET);

               usleep(1000000);//延迟
            }  
         }  
         else
         {
            slavetop(j);
            printf("slave again to op\n");
         }
      }
      else
      {
         printf("no slave on the bus\n");
      }
   }
   else
   {
      printf("no ethernet card\n");
   }
}
int main()
{
   printf("SOEM Start\n");
   char *argv = (char*)"enP3p49s0";
   simpletest(argv);

   printf("End program\n");
   return (0);
}`

closed 2023-12-31 by Mark9999999

nakarlsson, 2023-12-31:
The Wireshark say Working cnt : 1 , so your drive is probably in SAFEOP.

After you enter OP you sleep 300000, hence you stop sending processdata and the Watchdog will trip. That is not the way you do it, look at the test samples and other forum issues and you should be able to figure out how to do it.