OpenEtherCATsociety/SOEM #778: regarding COE mailbox sending and receiving issues

fyyxxm, 2024-01-03:
If I send a message through COE mailbox (`ecx-TxPDO()`), what should be the value of the input parameter `TxPDOnumber` for this function? Is it index or something else?

ArthurKetels, 2024-01-03:
As far as I know there are no slaves that support PDO over CoE. I never had the opportunity to test these functions.  It is in the EtherCAT standard however, that is why it is implemented in SOEM.

All PDO transfers are via virtual memory by means of FMMU. And the support for this lays in ecx_send_processdata, ecx_receive_processdata and friends. Or ,if you want to do it yourself, in low level instructions LRD, LWR and LRW.

fyyxxm, 2024-01-03:
Understood, thank you for your explanation

closed 2024-01-03 by fyyxxm