OpenEtherCATsociety/SOEM #781: SDO write failed

Yoouple, 2024-01-15:
Hi everyone! 
I encountered the problem of SDO communication failure when using the Panasonic MINAS A6 servo. 
The following is a detailed description:
In the Pre-OP state, the servo drive needs to be mapped to the object dictionary（Assigning PDO）, and the SDO write operation needs to be used in this step. This step uses the "ecx_SDOwrite" function, but the function sometimes returns failure.
Analysis of the "ecx_SDOwrite" function shows that after the command is written, there will be a readback operation to determine whether the command was written successfully.

      wkc = ecx_mbxsend(context, Slave, (ec_mbxbuft *)&MbxOut, EC_TIMEOUTTXM);
      if (wkc > 0)
      {
         ec_clearmbx(&MbxIn);
         /* read slave response */
         wkc = ecx_mbxreceive(context, Slave, (ec_mbxbuft *)&MbxIn, Timeout);

By grabbing the erroneous packets, it can be seen that after SDO communication is initiated, every time the slave station feedbacks a CoE packet of type "EMERGENCY", causing the readback determination of the "ecx_SDOwrite" function to fail.
[image]

I cannot confirm whether it is a problem with the Panasonic servo product itself or a problem with the SOEM library. The problem is sporadic and works fine most of the time.

Probably similar to what was mentioned in issue #764
> Also if drive works in OP state and SOEM app exits, drive fall in appropriate error, but after it I can't start it again - SDO write in PRE_OP fails. Only power reset helps.

This is the pcapng data captured when the problem occurred:
[Panasonic.zip](https://github.com/OpenEtherCATsociety/SOEM/files/13936191/Panasonic.zip)

I hope someone can bring me some inspiration. Thanks.

ArthurKetels, 2024-01-15:
This is slave specific behaviour. The slaves has a problem and issues an Emergency. This is stored in the ec_errorlist. But your application does not handle the issue and the slave refuses to honour the SDOwrite.

My suggestion is to parse the Emergency data, act on it and see if the slave then responds normally.

One emergency code I found in the capture was : 0x50 0xff 0x10 0x04
Perhaps you can look this up in the slave manual and see what it means.

Yoouple, 2024-01-16:
Thank you very much for your reply @ArthurKetels 

As you said, I also found this error code (0x50 0xff 0x10 0x04) in one of the EMERGENCY messages and found its meaning:
[image]
Looking at the servo's technical manual, it seems reasonable that this error code occurred, because before this the servo was working in the OP state and I terminated the SOEM application.

But what I don’t understand is that before this error code appears, there are a large number of EMERGENCY messages with empty content. I don't know what went wrong with the drive at this point.
[image]
However, the remapping of the object dictionary was successful, which shows that ecx_SDOwrite is effective, because I observed that the contents in 0x1C32\0x1C13\0x1A00\0x1600 are correct.

Once this error occurs, the real affected place is the "ecx_SDOread" function, because this function cannot read any meaningful data. This will affect the normal function of the **"ecx_readPDOmap"** function, causing the length calculation of SM2\SM3 to be incorrect, followed by an abnormal FMMU configuration.
[image]

It's not clear to me what triggered this error in this servo, but it does have a serious impact on the normal operation of the protocol stack.

By the way, when an error occurs, restart the servo drive or use the "ecx_config_init" function to rescan and the system will return to normal.

Yoouple, 2025-06-29:
> Hello engineer, I am using Panasonic Minas A6B motor and encountering the same error as you. When performing SODwrite, an 80.4 error signal is sent, and the status cannot be read. How did you solve this problem at that time? Looking forward to your reply.

Hi, after testing, I guess it is a bug with the drive. I tested the Minas A5B drive , and there is no problem.
The driver provides a parameter "10F3-05h" which can turn off emergency messages. You can refer to the Panasonic driver manual.
[Image.png]

Yoouple, 2025-07-06:
> Thank you for your reply,but Minas A6b servo also have function of 'clean error' .I tried to use this address to clear error.Even though the error has been cleared, the motor still cannot run,and I still couldnot write the byte.I 've given up developing with this motor in QT,But I will still be most greatful if you could provide some driving methods.

Hi, There are many possibilities that could cause this problem, and the CIA402 protocol manual may be helpful to you.