OpenEtherCATsociety/SOEM #785: Can't link wpcap.lib when using SOEM with cmake fetch_content

limymy, 2024-02-06:
MSVC is unable to link `wpcap.lib` when using this repository as a sub-project via the CMake `fetch_content` module. The error message is as follows:
```
[build] LINK : fatal error LNK1104: 无法打开文件“wpcap.lib” [E:\mycode\soem_test\build\cpp_test.vcxproj]
```
This issue may be related to the following lines in the CMakeLists.txt file:
https://github.com/OpenEtherCATsociety/SOEM/blob/1817b8b4fec10c66e4d97de73302ebe496bb82ce/CMakeLists.txt#L26-L30
It appears that the directory is not being passed to the parent project.

limymy, 2024-02-06:
I have submitted a pull request to solve this problem. #786

kj4tmp, 2024-04-29:
i believe I am hitting this same issue as part of my project that attempt to use `include_subdirectory` and `target_link_libraries` with SOEM

https://github.com/kj4tmp/pyecm
```
Building Custom Rule D:/a/pyecm/pyecm/CMakeLists.txt
      soem_ext.cpp
    D:\a\pyecm\pyecm\ext\SOEM\oshw\win32\nicdrv.h(21,10): error C1083: Cannot open include file: 'pcap.h': No such file or directory [D:\a\pyecm\pyecm\build\cp310-cp310-win32\soem_ext.vcxproj]
      (compiling source file '../../pyecm/soem_ext.cpp')
```

kj4tmp, 2024-04-29:
I was able to implement the following workaround by adding the following to my `CMakeLists.txt`

```
if(WIN32)
  # Add the directory containing pcap.h to the include directories
  # this is a hack to temporarily fix https://github.com/OpenEtherCATsociety/SOEM/issues/785
  target_include_directories(soem_ext PUBLIC
    ${CMAKE_CURRENT_LIST_DIR}/ext/SOEM/oshw/win32/wpcap/Include
  )
  if(CMAKE_SIZEOF_VOID_P EQUAL 8)
    target_link_directories(soem_ext PUBLIC ${CMAKE_CURRENT_LIST_DIR}/ext/SOEM/oshw/win32/wpcap/Lib/x64)
  elseif(CMAKE_SIZEOF_VOID_P EQUAL 4)
    target_link_directories(soem_ext PUBLIC ${CMAKE_CURRENT_LIST_DIR}/ext/SOEM/oshw/win32/wpcap/Lib)
  endif()
  
endif()

```
SOEM is a git submodule at `./ext/SOEM/` in my project

WARNING: I am not a cmake expert