OpenEtherCATsociety/SOEM #788: SOEM master slave communication

nehapagare17, 2024-02-09:
I have configured one linux based computer as master and other linux based computer as slave ,now i want to establish connection between both of them but i tried executing slaveinfo.c file build it in code block .but i am not getting executable file .what should be the next step and procedure to execute these file

ArthurKetels, 2024-02-09:
You forget to tell exactly what steps you have taken. SOEM is only an EtherCAT master library. It can not be configured as a slave. The standard way for a slave is to have a piece of dedicated hardware called the ESC (EtherCAT slave controller). It is possible to emulate the ESC in software, but I do not know of any open-source projects that implement that.

My suggestion is to read more about what EtherCAT is and how it works.

nehapagare17, 2024-02-10:
I have configured one PC as master using SOEM and other PC as slave using SOES. Now I have tried various steps to establish communication between them I have connected cables .but not getting where I am going wrong can you explain in brief how to make them communicating

External Email

ArthurKetels, 2024-02-10:
SOES will not implement a full EtherCAT slave. SOES is used to communicate with the ESC (see post above) for the higher level protocols. Without the proper hardware (or emulator) you can not have a working EtherCAT system. If you want to use a PC as slave you can use a device like this : https://www.beckhoff.com/en-en/products/ipc/pcs/accessories/fc1121.html 
As a side note, there is currently no driver in SOES for the fc1121.